ros2/rmw_fastrtps #14: Implement missing RMW functions

jacquelinekay, 2016-02-09:
The following functions in src/functions.cpp are not implemented yet:

https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L1770-L1806

We haven't exercised these functions outside of the ros1bridge yet, but this limitation came up recently in ros2/system_tests#103.

JaimeMartin, 2016-05-05:
Hi Jackie,

We are working on this. We will keep you posted.

jacquelinekay, 2016-05-05:
Thanks!

Pinging @wjwwood too since I believe he needs this for his work on the Graph API.

richiprosima, 2016-09-09:
service_is_available functionality for Fast-RTPS in #56.
@wjwwood I noticed that function wait_for_service calls only one time to rmw_service_is_available function. It should test more times, shouldn't it?

richiprosima, 2016-09-09:
#56

wjwwood, 2016-09-15:
@richiprosima `rmw_service_is_available` is a non-blocking call, so no it only checks once. The idea is that you're waiting for changes to the graph and only call this function once at the beginning and once each time the graph changes, until it either returns true or you've waited as long as you're willing to wait. See the implementation for `wait_for_service` in rclcpp:

https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/client.cpp#L65-L107

PR #56 (2016-09-20): Added new functionality: service_is_available
This PR contains the service_is_available functionality. To work properly Fast-RTPS has to be in/after commit eProsima/Fast-RTPS@24825bb0de6cc5fdf16949a350c40570f2b00f26. 

Closes #14.
closed 2016-09-20 by wjwwood