ros2/rmw_fastrtps #46: Repeated service calls fail (at least for parameters)

gerkey, 2016-07-01:
This test, which sets some parameters and then tries to retrieve them 10 times in a row:

https://github.com/ros2/system_tests/commit/0c7f5517f3a764683aae036ed97849d7632b826f

fails for FastRTPS, producing output like this:
```
[ RUN      ] test_local_parameters__rmw_fastrtps_cpp.local_synchronous_repeated
iteration: 0
iteration: 1
unknown file: Failure
C++ exception with description "failed to send request: cannot publish data, at /home/gerkey/ros2_ws_debug/src/eProsima/ROS-RMW-Fast-RTPS-cpp/rmw_fastrtps_cpp/src/functions.cpp:1619, at /home/gerkey/ros2_ws_debug/src/ros2/rcl/rcl/src/rcl/client.c:170" thrown in the test body.
```
That is, the first iteration works, but subsequent ones fail. The same test passes for OpenSplice (and, I'm pretty sure, Connext).

gerkey, 2016-07-06:
Interestingly, that repeated service call test also fails on Connext (http://ci.ros2.org/job/ci_linux/1629/testReport/(root)/projectroot/gtest_local_parameters__rmw_connext_cpp/):

```
[ RUN      ] test_local_parameters__rmw_connext_cpp.local_synchronous_repeated
[rmw|error_handling.c:93] error string being overwritten: C++ exception during construction of Requester, at /home/rosbuild/ci_scripts/ws/build/rcl_interfaces/rosidl_typesupport_connext_cpp/rcl_interfaces/srv/dds_connext/get_parameter_types__type_support.cpp:82
PRESContentFilteredTopic_createFilterProperty:!copy content filtered property "filter expression" field: reached maximum length for content filter property (current length: 262, max. length: 256). Please consider increasing contentfilter_property_max_length parameter under participant's resource limits.
PRESContentFilteredTopic_associateReader:!copy sequence for content filtered property data
DDS_Subscriber_create_datareader_disabledI:ERROR: Failed to associate reader and content filtered topic
DDSDataReader_impl::create_disabledI:!create reader
DDSDataReader_impl::createI:!create reader
DDSDomainParticipant_impl::create_datareader:ERROR: Failed to create datareader
initialize:!create DataReader
connext::details::EntityUntypedImpl::initialize:!failed (see previous errors)
unknown file: Failure
C++ exception with description "could not create client: failed to create requester, at /home/rosbuild/ci_scripts/ws/src/ros2/rmw_connext/rmw_connext_cpp/src/functions.cpp:919, at /home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl/src/rcl/client.c:84" thrown in the test body.
[  FAILED  ] test_local_parameters__rmw_connext_cpp.local_synchronous_repeated (3365 ms)
```
That one is for Linux; a similar result is produced for OSX (http://ci.ros2.org/job/ci_osx/1225/testReport/(root)/projectroot/gtest_local_parameters__rmw_connext_cpp/).

richiprosima, 2016-07-06:
@gerkey I was testing the test. The problem with FastRTPS is the History QoS is set to KEEP_LAST and depth is 10. This test sends more than ten samples and very fast. 11th sample writting fails because there are unacknowledged samples in the publisher history.
By default Heartbeat period is 5 seconds. Until Heartbeat is not sent, subscribers don't inform publisher they have the samples. To resolve this heartbeat period has to be decreased. I tested decreasing this period to 1ms.

```diff
diff --git a/rmw_fastrtps_cpp/src/functions.cpp b/rmw_fastrtps_cpp/src/functions.cpp
index e57f2ad..2b29954 100644
--- a/rmw_fastrtps_cpp/src/functions.cpp
+++ b/rmw_fastrtps_cpp/src/functions.cpp
@@ -1501,6 +1501,9 @@ fail:
         subscriberParam.topic.topicKind = NO_KEY;
         subscriberParam.topic.topicDataType = response_type_name;
         subscriberParam.topic.topicName = std::string(service_name) + "Reply";
+        // Response quickly to Heartbeats
+        subscriberParam.times.heartbeatResponseDelay.seconds = 0;
+        subscriberParam.times.heartbeatResponseDelay.fraction = 4294967;
 
         if(!get_datareader_qos(*qos_policies, subscriberParam))
             goto fail;
@@ -1517,6 +1520,9 @@ fail:
         publisherParam.topic.topicKind = NO_KEY;
         publisherParam.topic.topicDataType = request_type_name;
         publisherParam.topic.topicName = std::string(service_name) + "Request";
+        // 1 Heartbeat every 1ms
+        publisherParam.times.heartbeatPeriod.seconds = 0;
+        publisherParam.times.heartbeatPeriod.fraction = 4294967;
 
         if(!get_datawriter_qos(*qos_policies, publisherParam))
             goto fail;
@@ -1815,6 +1821,9 @@ fail:
         subscriberParam.topic.topicKind = NO_KEY;
         subscriberParam.topic.topicDataType = request_type_name;
         subscriberParam.topic.topicName = std::string(service_name) + "Request";
+        // Response quickly to Heartbeats
+        subscriberParam.times.heartbeatResponseDelay.seconds = 0;
+        subscriberParam.times.heartbeatResponseDelay.fraction = 4294967;
 
         if(!get_datareader_qos(*qos_policies, subscriberParam))
             goto fail;
@@ -1831,6 +1840,9 @@ fail:
         publisherParam.topic.topicKind = NO_KEY;
         publisherParam.topic.topicDataType = response_type_name;
         publisherParam.topic.topicName = std::string(service_name) + "Reply";
+        // 1 Heartbeat every 10ms
+        publisherParam.times.heartbeatPeriod.seconds = 0;
+        publisherParam.times.heartbeatPeriod.fraction = 4294967;
 
         if(!get_datawriter_qos(*qos_policies, publisherParam))
             goto fail;
```
Even so the test often fails. It is caused because FastRTPS can return sequence number 0 to user after writing a sample. I'm testing a solution of this bug in branch [hotfix/1736](https://github.com/eProsima/Fast-RTPS/tree/hotfix/1736)

dirk-thomas, 2016-07-06:
I guess the right fix is then to update the queue size in our test to accommodate all the messages sent in a short interval.

gerkey, 2016-07-06:
ros2/rclcpp#236 increases the history depth to 1000 (by using our already-defined parameter qos profile), which fixes the problems that I'm having.

@richiprosima, do you want to leave this ticket open while you keep investigating other fixes?

richiprosima, 2016-07-07:
@gerkey I pushed eProsima/Fast-RTPS@5a697cdae80. It solves the bug that often causes the return of a 0 sequence number. You can close this ticket if your test continues working successfully.

gerkey, 2016-07-09:
I don't know which change is the cause, but my tests are passing now, thanks!

closed 2016-07-09 by gerkey