ros2/rmw_fastrtps #48: repeated service response not received

dirk-thomas, 2016-07-05:
On of our tests (`test_client_scope_cpp` in `test_rclcpp`) creates a service client, calls the service and checks the response. After that it does the same cycle a second time. While it works fine on all other rmw implementation with FastRTPS the second response is never received. The server receives both requests and replies both time. But the client seems to never receive the second reply. Any help what we could do to narrow it down would be welcome. 

Btw. the issue happens on all platforms and build types (http://ci.ros2.org/view/nightly/job/nightly_linux_debug/lastCompletedBuild/testReport/).

wjwwood, 2016-07-05:
Is this related to @gerkey's report? https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/issues/46

His is about parameters, but it seems to be rooted in multiple service calls.

dirk-thomas, 2016-07-05:
That ticket mentioned an exception which doesn't happen in this case. I don't know if the source of the problem is related but the symptom is different. Also the case from this ticket is already on master and reproducible an all platforms and build types.

dhood, 2016-07-06:
I have been looking into this issue. that test started failing when https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/42 was merged: it's related to the QoS policy. before that PR was merged, every demo was essentially permanently using default parameters

For the `test_client_scope_cpp` test, now that the QoS policy takes effect, the server and the first client use the `RMW_QOS_PROFILE_SERVICES_DEFAULT` QoS profile. this uses `RMW_QOS_POLICY_RELIABLE` for the reliability setting: applied to both the [`datareader` (reply)](https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L1505) and [`datawriter` (request)](https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L1521).

the second client will not work with those same settings. the second client needs to have the datareader (reply) with `RMW_QOS_POLICY_RELIABLE` and the datawriter (request) with `RMW_QOS_POLICY_BEST_EFFORT`.

therefore that test can be fixed by leaving the reliability setting as `RMW_QOS_PROFILE_SYSTEM_DEFAULT` which will result in the aforementioned settings being applied. (we do not currently have capabilities to set the reliability settings separately for the datawriter and datareader from what I see). it should only be applied to fastrtps or it will break connext (I haven't checked with opensplice) https://github.com/ros2/system_tests/commit/c460537948b7ef01120bd8aedf6f4161d2743055 http://ci.ros2.org/job/ci_linux/1634/

For the first client, both the datareader and datawriter are using `RMW_QOS_POLICY_RELIABLE`, but it also works if you use `RMW_QOS_PROFILE_SYSTEM_DEFAULT` (which assigns reliable to datareader and best effort to datawriter). for the second client there is no choice.

I do not yet understand why this is the case. 

it also seems that PR https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/pull/44 fixes the behaviour of this test, so it might have to do with particular combinations of reliability and durability settings combined.

dhood, 2016-07-06:
update: the claimed default durability settings for publishers [here](https://github.com/eProsima/Fast-RTPS/blob/3cf7996c9cea21a4bdc4ee5f8dc6763069cd85f2/include/fastrtps/qos/QosPolicies.h#L60) are not being applied (they're getting subscriber defaults). working on a fix

dhood, 2016-07-07:
So, this issue can be fixed by any combination of the following:

1. changing reliability to system defaults [(original fix)](https://github.com/ros2/system_tests/commit/c460537948b7ef01120bd8aedf6f4161d2743055)
2. changing durability to volatile (#44)
3. changing fastrtps defaults back to how they were before [this commit](https://github.com/eProsima/Fast-RTPS/blob/93088db71c088dc04ffa77418c2492d5012b8b13/src/cpp/qos/WriterQos.cpp)

1 was a bit of a hack. since #44 will mask fix this issue I will add another test to keep track of this issue.

richiprosima, 2016-07-29:
@dirk-thomas  @dhood @wjwwood Last update of FastRTPS should be solved this bug.

dirk-thomas, 2016-07-29:
@richiprosima Can you please reference the change which addresses the problem.

richiprosima, 2016-08-05:
Last commit adding the send of an initial heartbeat and nack is eProsima/Fast-RTPS@7b9fdd29b77883d5310e82e454dcef0fb9973b41

dhood, 2016-09-21:
Thanks @richiprosima, can confirm that this is fixed. This can be closed.

closed 2016-09-21 by dirk-thomas