ros2/rmw_fastrtps #64: missing functionality in graph API

wjwwood, 2016-10-29:
While working on local graph changes for Connext I noticed two things about the FastRTPS implementation:

- publishers and subscribers do not get removed from topic_names_and_types ever (even when destroyed)
- count publisher and subscribers does not appear to count the publishers and subscribers but rather the number of different types on a topic

This can be shown using the graph tests in `rcl`, which are currently disabled for Fast RTPS since they don't pass:

https://github.com/ros2/rcl/commit/596c0ec9faaa81cb4ba9dc064c27ba340412c023#diff-e4167ceebe2946426e927f55e4392af4R66

I didn't want to hold up the other PR's to fix this. I spent a few minutes looking at it, and I think the first issues could be addressed by storing the publisher/subscriber GUID's. The second issue will require being able to tell the difference in a publisher or subscriber coming up versus going down. I wasn't able to figure out how to do that and I wasn't even sure there was any notification that they went down at all.

You can see here that the `topicNtypes` map is inserted into:

https://github.com/eProsima/ROS-RMW-Fast-RTPS-cpp/blob/master/rmw_fastrtps_cpp/src/functions.cpp#L353

But as far as I can see nothing is ever removed. Maybe someone can give us a lead on how to accomplish this using the data coming into that function. You can see similar, but slightly different code for how we accomplish this in Connext:

https://github.com/ros2/rmw_connext/blob/master/rmw_connext_shared_cpp/src/shared_functions.cpp#L78-L82

dirk-thomas, 2017-06-02:
Can this be closed now?

wjwwood, 2017-06-02:
Yup, thanks.

closed 2017-06-02 by wjwwood

dhood, 2017-06-05:
for future reference the relevant PR is https://github.com/ros2/rmw_fastrtps/pull/111