ros2/rmw_fastrtps #68: Error when publishing more than 5000 messages and using "keep all" history

dhood, 2016-11-29:
**Context**:
I am leaving a publisher with default QoS settings to run, without any subscriptions. After publishing 5000 messages, I am getting a "[Maximum number of allowed reserved caches reached](https://github.com/eProsima/Fast-RTPS/blob/24825bb0de6cc5fdf16949a350c40570f2b00f26/src/cpp/rtps/history/CacheChangePool.cpp#L229)" warning from Fast RTPS, which leads to a [runtime error in `rmw_fastrtps_cpp`](https://github.com/ros2/rmw_fastrtps/blob/8c42e3807523d4422dd4e5c731e824b3cea9f92c/rmw_fastrtps_cpp/src/functions.cpp#L921).

**To replicate**:
Run a publisher [like this one](https://github.com/ros2/examples/blob/895843476f9a3f1180ac7d373fec48fe88ca8141/rclcpp/topics/minimal_talker/talker_timer_lambda.cpp) (using rclcpp or rclpy) alone and let it run until it gets to 5000 messages. Increasing the frequency helps speed things up but this happens even at 1Hz.

**Outcome**:
This is the output with logging enabled on Fast RTPS:
```
Publishing: [Hello, world! 4999]
[PUBLISHER Info] Writing new data -> Function write
[RTPS_WRITER Info] Creating new change -> Function new_change
[RTPS_HISTORY Info] Change 5000 added with 23 bytes -> Function add_change
[RTPS_WRITER Info] No reader proxy to add change. -> Function unsent_change_added_to_history
Publishing: [Hello, world! 5000]
[PUBLISHER Info] Writing new data -> Function write
[RTPS_WRITER Info] Creating new change -> Function new_change
[RTPS_UTILS Info] Allocating group of cache changes of size: 510 -> Function allocateGroup
[RTPS_HISTORY Warning] Maximum number of allowed reserved caches reached -> Function allocateGroup
[RTPS_WRITER Warning] Problem reserving Cache from the History -> Function new_change
terminate called after throwing an instance of 'std::runtime_error'
  what():  failed to publish message: cannot publish data, at /home/dhood/ros2_ws1/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/functions.cpp:928, at /home/dhood/ros2_ws1/src/ros2/rcl/rcl/src/rcl/publisher.c:131
Aborted (core dumped)
```

**Further details**:
1. When a subscription is subscribed to the topic, the error does not occur. When the subscription terminates, the error from the publisher will then happen 5000 messages later.
1. **Changing the QoS history setting to "keep last" will prevent this error from happening.**
1. Changing the QoS durability to volatile does not have any effect.
1. Changing the QoS depth to a smaller (1) or larger (6000) number does not have any effect.
1. Changing the QoS reliability to best effort does not have any effect.

mikaelarguedas, 2016-11-29:
It seems that it's a limitation on the maximum size of the history cache in FastRTPS so it seems coherent that changing the history settings is the only thing that fixes it.
What would be the proper fix? changing the behavior of "keep all" to "keep 5000"? or monitor it and send a proper error when Fast-RTPS reaches it's history cache size limit ?

dhood, 2016-11-29:
from speaking to @wjwwood I understood that with the default QoS profile, the history cache limit shouldn't be being reached in the first place (for a queue size less than 5000)

@wjwwood is this accurate?
**Expected behaviour**:
1. Publishing without a subscription to the topic should store either nothing in the cache (in the case of "volatile" durability) or at most `queue_size` samples in the cache (in the case of "transient_local" durability). (I believe the default in fast rtps to be ["transient local" for publishers](https://github.com/eProsima/Fast-RTPS/blob/24825bb0de6cc5fdf16949a350c40570f2b00f26/include/fastrtps/qos/QosPolicies.h#L60)).
1. Publishing with a subscription to the topic should store at most `queue_size` samples in the cache, and if they are not being removed from the cache fast enough then either overwrite them (in the case of "best effort" reliability) or block (in the case of "reliable" reliability, because otherwise you would have to overwrite something, and then "keeping all" wouldn't be respected).

Either way, this error should only occur if the `queue_size` is >5000.

The proper fix isn't clear at this point (at least not from my point of view) as it will need some investigation into why the expected behaviour isn't being observed

mikaelarguedas, 2016-11-29:
Interesting, thanks for the more detailed explanation

wjwwood, 2016-11-29:
I think what you've described is what I was talking about. The thing that is weird to me is that changing it to volatile doesn't eliminate the issue (among the other weird cases you pointed out, like changing to best effort).

dhood, 2016-11-29:
that's correct. I won't look into this any further for now, but I suppose the data is being added to the queue inappropriately when there are no subscriptions, and therefore it's then not being removed because it's not being ACK/NACKed.

This affects any users that want to use the default qos profile and leave publishers running without subscriptions (which is potentially a lot of use cases), so we should come back to this issue and/or increase its visibility before the beta.

dhood, 2016-12-02:
I have been reading Connext and Opensplice documentation and think there are these changes to be made to our understanding of how the system should work:
a. the queue size does not apply in any way for "keep all" durability. [it is ignored.](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/HISTORY_QosPolicy.htm#sending_2410472787_1087838)
b. while reliable publishing can block, it happens [if the resource limits are exceeded, not because the queue is full](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/api/connext_dds/api_cpp/structFooDataWriter.html#abb3770f202340bc819368463987eb055)

So I think that this is more appropriate:
**Expected behaviour:**
1. Publishing without a subscription to the topic should:
    1. Store nothing in the cache (in the case of "volatile" durability). ([Connext API says](https://community.rti.com/static/documentation/connext-dds/5.2.3/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/DURABILITY_QosPolicy.htm#sending_2410472787_1086842) "DDS_VOLATILE_DURABILITY_QOS: **Do not save** or deliver old DDS samples.")
    1. Store at most `queue_size` samples in the cache (in the case of "transient_local" durability + "keep last" history).
    1. Store everything in the cache (in the case of "transient_local" durability + "keep all" history).
1. Publishing with a subscription to the topic should (in the case of either "best effort" or "reliable" reliability):
   1. Store at most `queue_size` samples in the cache, and if they are not being removed from the cache fast enough then overwrite them (in the case of "keep last" history).
    1. Store all that's physically possible in the case of "keep all" history.

The part that's not behaving as expected is 1.i. It is giving the behaviour of 1.iii.

While the durability settings have been added to `rmw_fastrtps_cpp` more recently than the others (in https://github.com/ros2/rmw_fastrtps/pull/44), I can confirm that they seem to be being applied correctly: a "transient_local" subscription will fail to connect to a "volatile" publisher with `[RTPS_EDP Warning] INCOMPATIBLE QOS` as expected.

It therefore seems that Fast RTPS doesn't distinguish between case 1.i and 1.iii as we are expecting it to. It would be good to show this with one of their examples and see what they think. In the meantime we can work around it with "keep last."

dhood, 2016-12-08:
Update: I haven't been able to reproduce this [using the Fast RTPS "HelloWorld" publisher](https://github.com/eProsima/Fast-RTPS/compare/master...dhood:cache-full) so I think there is a subtle difference somewhere between how we configure the datawriter and how they do.

richiprosima, 2017-02-08:
Sorry for commenting so late. I didn't be aware of this issue. This issue should be resolved after eProsima/Fast-RTPS@75b24b45a197afaea8e759dabecc89daaca74a85. This commit fixes a bug related to your problem.

dirk-thomas, 2017-02-08:
@richiprosima The link to the referenced commit doesn't work. Can you comment with a link to the change.

richiprosima, 2017-02-09:
The commit was not introduced on master branch yet. Right now it is. Sorry

dirk-thomas, 2017-07-14:
@dhood Can this be closed?

dhood, 2017-07-14:
fixed thanks!

closed 2017-07-14 by dhood