ros2/rmw_fastrtps #70: Limitation size of String

Theosakamg, 2016-12-04:
from last issue #60 with msg String limitation

_base on https://github.com/rosalfred/smarthome_media_msgs/tree/ros2_ 

make services for :
- **MediaGetItem** (array of MediaItem) https://github.com/rosalfred/smarthome_media_msgs/blob/ros2/srv/MediaGetItem.srv#L3
- **MediaGetItems** (only once MediaItem) https://github.com/rosalfred/smarthome_media_msgs/blob/ros2/srv/MediaGetItems.srv#L5

with MediaItem composed by a data field of type String https://github.com/rosalfred/smarthome_media_msgs/blob/ros2/msg/MediaItem.msg#L6

when I call **MediaGetItems** and return a Array of MediaItem with data field setting with String of 1024 characters, all work perfectly.
if i call **MediaGetItem** and return only once MediaItem with data field setting with String of 256 characters all work,

but if i call  **MediaGetItem** and send 1024 characters i recieve a publisher error...

_I tested with Python and Java node on Linux._

Theosakamg, 2016-12-04:
code of service for test :
```
import rclpy

from smarthome_media_msgs.srv import MediaGetItem
from smarthome_media_msgs.msg import MediaType

def callback(request, response):
    response.item.mediaid = 0
    response.item.mediatype.value = MediaType.VIDEO_MOVIE

    # Not working !!! (for MediaGetItem, but work on MediaGetItems)
    #response.item.data = "IGSCJsNjNyZE1GI9Kh6MBUkDOqwZahGVjXo0hfa0TDSNiFJ09bfjBcTmBNJBwsAro46khYIvF3UH4n9H2krBCFYqiGOMEBqnBUt75vIz0XlV85aehXvVkFYptKfmbm4Rn1kgXJKQge3XIFYRLDWR84nI3Jx4zFe5kZiMgyW1r0AvuLoxtCmErnNaoVYj0TtzWztJBLLR9LDvVEE2lQ6yM1pkosCqpi2f9v7TVVIWOnUGJP7HLeesEEW9CoyZ9fap2xxfUwbqC5a83JYRGNTMmzJhQsqvbvNTXXNs5eDKE8y426nGaIbO6INe8VAVlpXCrqWuoTFuQmvLM3R0eCcSaW3Nv70ZYscTmIgJqyNZHRm7gxUXPqfCAP8YgxKnNI8OXWOV801RQ3JoyY8xR3DZrDUIHw4ypzeyRIqYtlWT96gV6X07aRoGUW9itpML8H66Offj8fCLu6aY7fb7BltRy3ky60X4TE1aBsk9wftb59yGAzPxBMjtYHytkTLeWjuw5ycwUmBBjyEAhAYoszDNnsjGS6UAtO92qxouKau0up8WsVtGbkS39TFE1mAMNwXI3RZ3CP3LajtKLMWjO9N7lG7slLwXyyPOhq17j12ok15C4xIs2aeZQQDXrxX3RJsclM5veoC7RtCcualrK39a87D12Ctxzwk87HUbu5w4YgU0Kp4p3cuGwioHwQxNGmnUzuhElnirKTrawLzN5Krytfr8gPSIVzaQHECYW0cEbW3boht7ReZaMUCrLg2P4WBSzGcxPPzXu4tq8ATr6X7uhmte6SeExl7ZeGLPS6mpOLU2eF2FgeUFobiIuviVBqtcnuLU9vDNzPYMSVbWV0UEy7uDICrXfQONX8ixmKigDQ8Pgz6USBBiuGnPpsl5A6gkeBbmCmY4zYGiwGtCuMeDNNNDMVJlunNb1YSThMeZ8no9hZczPsJVY9ALarQJXTROvEVAv5Da3DHysfE0WcJj1v4HqN2ErFvl99ptehU46jAYW1qZ7KGZcbpC5Co0mNp2"
    # Working !!!
    response.item.data = "azerty"

    print('Incoming request\n')
    return response

def main(args=None):
    rclpy.init(args)

    node = rclpy.create_node('test_server')

    srv = node.create_service(MediaGetItem, 'get_item', callback)
    max_iter = 3
    i = 0
    while rclpy.ok() and i < max_iter:
        rclpy.spin_once(node)
        i += 1
    node.destroy_service(srv)
if __name__ == '__main__':
    main()
```

mikaelarguedas, 2016-12-05:
Bug confirmed on Fast-RTPS only, failing for strings longer than 256 characters regardless of the typesupport used (c or c++). As you pointed out it works fine when the long string is an array element.
Thanks for reporting it

dirk-thomas, 2016-12-05:
But [this test](https://github.com/ros2/system_tests/blob/b3969f43d2d9ee1931b48e170ca00b27146ccec2/test_communication/test/message_fixtures.hpp#L117) is currently passing. So I don't think it is a general problem of long strings.

mikaelarguedas, 2016-12-05:
Yes as reported in the issue, messages with a long `string` pass but a request/reply with a `string` field longer than 256 fails to be published. The tests added in https://github.com/ros2/system_tests/issues/179 fail for both `rclcpp` and `rclpy` so it must be a bug in the `rmw_implementation`.
I agree that it's not a problem of `string` themselves, especially because as mentioned by @Theosakamg a long request or reply with the same string in a nested message is published successfully.

Theosakamg, 2016-12-05:
Just a precision, I detected that on my machine (x86_64) the message passed if lower than 409 characters (not 256).

Theosakamg, 2016-12-06:
@mikaelarguedas your branch fix_long_req_reply seem fixed pb ! :+1:

mikaelarguedas, 2016-12-06:
Awesome! I'm still working on it but this should be in master in the next few days

Theosakamg, 2016-12-07:
Perfect !

closed 2016-12-07 by Theosakamg