ros2/rmw_fastrtps #72: Magic_numbers in calculateMaxSerializedSize

mikaelarguedas, 2016-12-07:
This is a follow-up of https://github.com/ros2/rmw_fastrtps/issues/36
There are still magic numbers ([here](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport_impl.h#L816) and [here](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport_impl.h#L834)) use to determine the maximum size of a sample.
It would be good to investigate if we can get rid of them and improve the logic of the [MaxSize calculation](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/MessageTypeSupport_impl.h#L41-L42)

allenh1, 2017-06-01:
@mikaelarguedas This might also be something I could work on

mikaelarguedas, 2017-08-04:
@allenh1 Is this one still on your radar ?

allenh1, 2017-08-04:
Yep -- I was waiting on the refactor to finish, but looks like I can go after it now.

mikaelarguedas, 2017-09-01:
There is now only [one magic number left](https://github.com/ros2/rmw_fastrtps/blob/79d8dc4ed25306ce44d97fede4d8fa0f474f7a34/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport_impl.hpp#L427).

@richiprosima I see there is a [todo in your name](https://github.com/ros2/rmw_fastrtps/blob/79d8dc4ed25306ce44d97fede4d8fa0f474f7a34/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport_impl.hpp#L443) to modify it, do you have any insight as of if this variable should be kept ?

richiprosima, 2017-09-04:
That value is inherited from when there was only preallocated memory policy in FastRTPS, and dynamic types had to have a maximum length. For sequences/arrays the maximum length was 100. Now ROS2 uses preallocated_with_realloc memory policy and maximum lengths are not needed.

dirk-thomas, 2017-09-04:
@richiprosima can you please propose a patch how to get rid of this variable. It isn't clear to us how to remove it cleanly.

richiware, 2017-09-07:
I'm working on it

richiware, 2017-09-11:
PR #152 removes those magic numbers.

mikaelarguedas, 2017-09-14:
All magic numbers have now been removed :tada: closing

closed 2017-09-14 by mikaelarguedas