ros2/rmw_fastrtps #90: FindFastRTPS.cmake uses the env var FASTRPCHOME instead of FASTRTPSHOME

jespersmith, 2017-02-14:
In FindFastRTPS.cmake:

if($ENV{FASTRPCHOME})
    set(FastRTPS_HOME $ENV{FASTRPCHOME})
endif()

This should of course be

if($ENV{FASTRTPSHOME})
    set(FastRTPS_HOME $ENV{FASTRTPSHOME})
endif()

dirk-thomas, 2017-02-14:
@jespersmith Can you please create a PR with the suggested patch.

@richiprosima Can you please comment on it since you wrote this originally.

jespersmith, 2017-02-14:
@dirk-thomas No. I'm using this file in another project, just pointing out the error in the documentation.

dirk-thomas, 2017-02-14:
@jespersmith You can edit the file through the GitHub web interface and it will automatically create a fork for you, add a commit with your change and create a pull request for it. We can then trigger a CI build for the PR to ensure it continues working (which I would expect).

jespersmith, 2017-02-14:
@dirk-thomas That's easy. See pull request #91

dirk-thomas, 2017-02-14:
@jespersmith Thank you!

PR #110 (2017-05-18): remove obsolete stuff from CMake module
Replaces #91. Fixes #90.

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]
closed 2017-05-18 by dirk-thomas