ros2/rmw_fastrtps #92: Enabling FastRTPS' ThroughputController

dhood, 2017-02-28:
This was originally discussed in https://github.com/ros2/rmw_fastrtps/issues/36
It can be enabled by un-commenting [these lines](https://github.com/ros2/rmw_fastrtps/blob/d353b3ce34849147b0bddcc184810b04cebfb736/rmw_fastrtps_cpp/src/functions.cpp#L812)

**Motivation:**
- Enhance performance for sending large messages with "best effort" reliability (otherwise all message fragments are sent at the same time, collide, and are not resent)

**Reasons against:**
- This can limit the throughput of systems.

**Options:**
1. Enable only if best effort is being used. If users want higher throughput than the controller they will have to use reliable reliability.
1. Enable always and accept the limited throughput.
1. Expose a flow controller QoS setting, as it's also available in Connext [see this demo](https://community.rti.com/examples/asynchronous-publisher)
1. Expose the appropriate symbols to the user so that they can set the throughput controller themselves.

If we decide to go for 1 or 2, it may be useful to know that for the image_tools demo sending 320x240 @30fps we send ~8million bytes per second. A throughput controller of ~30,000 bytes/10ms (3million/s) is sufficient.

dhood, 2017-02-28:
Thanks @mikaelarguedas for pointing out that implementing the throughput controller by un-commenting [these lines](https://github.com/ros2/rmw_fastrtps/blob/d353b3ce34849147b0bddcc184810b04cebfb736/rmw_fastrtps_cpp/src/functions.cpp#L812) will apply the same controller to all topics which will not be sufficient for cases where you need different controllers for different topics.

mikaelarguedas, 2017-08-04:
@dhood do you think that this can be closed in favor of https://github.com/ros2/rmw/issues/115 ?

sloretz, 2018-03-08:
I removed `more-information-needed` because it seemed to be added in reference to ros2/rmw#115 which itself was closed in favor of more specific tickets, so this ticket shouldn't be closed in favor of that one.

It seems like the milestone for this could be untargeted - it is desirable but no one appears to be advocating for it to be resolve anytime soon.

dirk-thomas, 2018-03-08:
I changed the milestone to `Bouncy` since I think it is crucial to fix the performance problem we are facing in e.g. the image demos with FastRTPS in the past releases.

mikaelarguedas, 2018-06-18:
With https://github.com/ros2/rmw_fastrtps/issues/202 (and this commit https://github.com/eProsima/Fast-RTPS/commit/201a393b500a35d56d3e46b5b6c2c6a96e85928e) it looks llike the performace issues we were facing with Fast-RTPS have been resolved. As the performance is now similar to the other RMW implementations I'm going to close this issue

closed 2018-06-18 by mikaelarguedas

paulbovbel, 2018-06-18:
Just to verify since https://github.com/ros2/rmw/issues/115 is closed as well, does this mean QoS settings for flow control will not be exposed via RMW?