ros2/rmw_fastrtps #93: Best effort + fragmentation + OS X = no images

dhood, 2017-03-01:
@richiprosima @JaimeMartin, we are trying to understand why particular conditions impact the reception of best effort data.

For high-bandwidth data, we have found a difference in performance on different operating systems. For example, with a **single publisher** publishing images (320x240 image x 1 byte/channel x 3 channels @ 30fps) and receiving them **on the same machine** with "best effort" reliability, it will work fine on linux, but not at all on OS X (no images are received).

1. Using a throughput controller, as you have suggested in https://github.com/ros2/rmw_fastrtps/issues/81 and https://github.com/ros2/rmw_fastrtps/issues/36, will fix the performance on OS X, but we are trying to understand (1) why there is a difference between operating systems and (2) if there is an alternative solution, since enabling the throughput controller will limit the rate at which users can receive data.

1. Instead of a throughput controller, disabling multicast on subscribers as suggested [in this comment](https://github.com/ros2/rmw_fastrtps/issues/36#issuecomment-228753919) improves performance but only slightly.

1. Surprisingly, performance can be improved even without a throughput controller or disabling multicast, if the publishing frequency is increased from 30fps to 100fps. At 30fps, 0 messages are received, but at 100fps, ~20% of messages are received, at 250fps, ~40% of messages are received...

Do you have an idea of what causes the messages to have low reception rates even with a single sender, and the receiver on the same machine?
@wjwwood has suggested that perhaps "the kernel's udp buffer is getting overwhelmed by the bursty nature without the flow controller" but you may have more insight given your experience.

Do you know why reception rate improves with higher publishing frequency? Thanks in advance.

---

To reproduce using our [image demo](https://github.com/ros2/ros2/wiki/Quality-Of-Service) (on OS X):
```
$ cam2image -b -r 0 -f 30
```

```
$ showimage -s 0 -r 0
```

richiprosima, 2017-03-02:
@dhood I was able to reproduce it in a VM. I will try to identify the problem. Thanks.

mikaelarguedas, 2017-08-04:
@dhood & @richiprosima any update on this ?

richiprosima, 2017-08-04:
I've tested again in my VM with last version of Fast RTPS and it has good performance. Maybe @dhood may check it too.

mikaelarguedas, 2017-08-04:
Thanks @richiprosima for the prompt response!
I'll leave it to @dhood to test and comment or close this issue accordingly

dhood, 2017-08-04:
yep looks fixed thanks!

closed 2017-08-04 by dhood