ros2/rmw_fastrtps #95: Using non-default FastRTPS heartbeat period

dhood, 2017-03-08:
Spin-off thread from https://github.com/ros2/rmw_fastrtps/issues/81 to capture the discussion on using a FastRTPS heartbeat period other than the default, originally suggested in https://github.com/ros2/rmw_fastrtps/issues/36.

This can be configured by un-commenting [these lines](https://github.com/ros2/rmw_fastrtps/blob/d353b3ce34849147b0bddcc184810b04cebfb736/rmw_fastrtps_cpp/src/functions.cpp#L808..L809).

**Motivation:**
Enhance performance for sending large messages with "reliable" reliability. The heartbeat from the publisher is used to get ACKNACKs from the subscriber. With the default heartbeat period ([3s according to this](https://github.com/eProsima/Fast-RTPS/blob/5301ef203d45528a083821c3ba582164d782360b/include/fastrtps/rtps/attributes/WriterAttributes.h#L56)), when something prevents messages/message fragments from being received, the subscriber can't communicate this with the publisher at a high enough frequency, so data is not re-sent often enough to have high quality connections.

**Reasons against just uncommenting the lines:**
- As @mikaelarguedas [pointed out](https://github.com/ros2/rmw_fastrtps/issues/81#issuecomment-281540688): This will affect all connections with "reliable" reliability, and has potential to flood the network in large-scale systems.

**Options:**
1. Reduce heartbeat period always and accept the overhead.
1. As @mikaelarguedas [suggested](https://github.com/ros2/rmw_fastrtps/issues/81#issuecomment-281540688): Reduce heartbeat period only for particular QoS profiles, e.g. those that are commonly used for large messages. This could be an existing profile or a new one that is used for large messages with reliable reliability.
1. Expose the heartbeat as a QoS setting, as it is also configurable in Connext [(through the datawriter protocol QoS)](https://community.rti.com/static/documentation/connext-dds/5.2.0/doc/manuals/connext_dds/html_files/RTI_ConnextDDS_CoreLibraries_UsersManual/Content/UsersManual/DATA_WRITER_PROTOCOL_Qos.htm#sending_2410472787_1093377)
1. Expose the appropriate symbols to the user so that they can set the heartbeat period themselves.
1. Wait for eProsima to implement piggyback heartbeats, which will increase the number of heartbeats sent with more data being sent ([ETA pending](https://github.com/eProsima/Fast-RTPS/issues/79)).

In the meantime the fix for users that need improved performance is to change the heartbeat period build `rmw_fastrtps` from source, manually configuring a reduced heartbeat period.

dhood, 2017-03-08:
My understanding is that even with small data, it could be up to 3s before a reliable subscriber requests re-sending of unreceived data. If someone can double check my understanding on this, then I think the motivation is, more broadly, to decrease the latency for reliable connections in the event that data has to be re-sent. This is relevant even for topics sending small amounts of data infrequently (if the network is bad).

jespersmith, 2017-04-12:
Could a workaround for now be to provide env variables or configuration options so it can be set per-process?

JaimeMartin, 2017-04-12:
Hi guys,

We have mostly finished a new feature to setup the QoS parameters through an XML file in the app folder. That will solve this issue.

mikaelarguedas, 2017-04-12:
I also heard that eProsima (feel free to correct / complete this statement @JaimeMartin) will implement piggyback heartbeat soon. That would solve the problem of sending big amount of data without having to tweak heartbeat period manually on a per process / per topic basis.

JaimeMartin, 2017-04-12:
Also true. We are covering every angle of sending large data.

Regards,
Jaime Martin Losa
CEO eProsima - The middleware experts

Phone: + 34 607 91 37 45
Twitter: @JaimeMartinLosa
www.eProsima.com

dirk-thomas, 2017-07-14:
@dhood Can this be closed?

mikaelarguedas, 2017-07-14:
I think that the question of exposing the heartbeat period through QoS settings in ROS is still unanswered at the moment. This could be closed in favor of a global thread about "exposing DDS symbols through the interface" v.s. "What additional QoS settings we want to expose through the ROS interface"

mikaelarguedas, 2017-08-04:
closing this in favor of https://github.com/ros2/rmw/issues/115

closed 2017-08-04 by mikaelarguedas