ros2/rmw_fastrtps #96: Segfault on raspberry pi

firesurfer, 2017-03-28:
Hi,
I created a test that transfers a rather big messsage (around 8Mb) via the Fast-RTPS middleware. The data is transferred between two raspberry pis running the latest version of ros2 (master branches) and also the master branch of Fast-RTPS. The network in between contains multiple switches that are configured to use rstp. When I disconnect a cable from a switch the stp tree is rebuilded by the switches and the data is transferred via another connection. This works in most cases. How ever I managed to crash the talker by reconnecting the first network cable. (See segfault attached)

Talker:
https://gist.github.com/firesurfer/1755157a8bb2ada6816301f1ef50c755
Listener:
https://gist.github.com/firesurfer/26bdd1c3ed29b96f71d5c58720d08ca5

```
Thread 8 "big_packet_test" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x72eff450 (LWP 3236)]
0x766f8794 in eprosima::fastrtps::rtps::ReaderProxy::set_change_to_status(eprosima::fastrtps::rtps::CacheChange_t const*, eprosima::fastrtps::rtps::ChangeForReaderStatus_t)
    () from /home/pi/workspace/ros2_ws/install/lib/libfastrtps.so
(gdb) backtrace 
#0  0x766f8794 in eprosima::fastrtps::rtps::ReaderProxy::set_change_to_status(eprosima::fastrtps::rtps::CacheChange_t const*, eprosima::fastrtps::rtps::ChangeForReaderStatus_t) () from /home/pi/workspace/ros2_ws/install/lib/libfastrtps.so
#1  0x766f3978 in eprosima::fastrtps::rtps::StatefulWriter::send_any_unsent_changes()
    () from /home/pi/workspace/ros2_ws/install/lib/libfastrtps.so
#2  0x766ee630 in eprosima::fastrtps::rtps::AsyncWriterThread::run() ()
   from /home/pi/workspace/ros2_ws/install/lib/libfastrtps.so
#3  0x76a6693c in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
#4  0x76ebfff4 in start_thread (arg=0x72eff450) at pthread_create.c:335
#5  0x768b1b10 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:86
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

codebot, 2017-03-28:
Thank you for this detailed bug report and stack trace. Just from the stack, I'm guessing that the issue lies in FastRTPS, so I would suggest also filing this over with the eProsima folks: https://github.com/eProsima/Fast-RTPS

firesurfer, 2017-03-28:
See https://github.com/eProsima/Fast-RTPS/issues/84

firesurfer, 2017-04-11:
I haven't got any reaction form the FastRTPS vendors in the last 14 days. Just wanted to ask if this is normal?

dirk-thomas, 2017-04-11:
@richiprosima ping.

firesurfer, 2017-04-18:
@JaimeMartin 
This example https://github.com/firesurfer/fastrtps_segfault_demo produces the segfault in an environment as described in the entry post.

richiprosima, 2017-04-21:
With ros2/ros2@0f0d0614f0a1313a7ac44f7d30651421453f7f9e ROS2 returned to follow Fast-RTPS master branch 8 days ago. I think your segmentation fault is related to eProsima/Fast-RTPS#78 which was fixed on eProsima/Fast-RTPS@ad59a6cf29e6f15a105193ea304cff645f0b9137. Can you test with lastest master branch of ROS2?

Thanks

firesurfer, 2017-05-23:
I just found some time to test this bug again on our testing setup. Couldn't reproduce it so far.

closed 2017-05-23 by firesurfer