ros2/rmw_fastrtps #98: RTPS_WRITER error when publishing large amount of data

sidd1, 2017-04-03:
Hi! I'm running into trouble when publishing large amounts of data. Without a subscriber, the node runs fine without any errors. But if I have a second node subscribing to this topic, I run into a bunch of errors on the publisher end. - 

    [RTPS_WRITER Error] Problem adding DATA_FRAG submsg to the CDRMessage, buffer too small -> Function prepareDataFragSubM
    [RTPS_WRITER Error] A problem occurred when adding a message -> Function send_Changes_AsData

And my Subscriber seg faults with this error - 

    terminate called after throwing an instance of 'std::out_of_range'
    what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)

The behaviour remains the same irrespective of having a c++ or python subscriber.

wjwwood, 2017-04-03:
What version of ROS 2 are you using?

sidd1, 2017-04-03:
I'm using the release-beta1 version (Cloned the repositories from `https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos`).

wjwwood, 2017-04-03:
Can you estimate the size of the message in bytes and/or tell us what the message type is?

Also, I know it might be difficult for you do this, but could you consider trying master? The repos file is at https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos. It uses a newer version of Fast-RTPS which may have addressed some of these issues already. It would be nice to avoid digging into the issue if it's already fixed on the latest version. You could also try the latest packaging binary:

- http://ci.ros2.org/view/packaging/job/packaging_linux/632/artifact/ws/ros2-package-linux.tar.bz2

If you're unable to try out master then we'll have to spend some time trying to reproduce it. A minimal example would help us get to that quicker.

sidd1, 2017-04-07:
Sorry for the delay. The initial bug I reported was caused on a VM I was running. Trying the same piece of code on a physical machine, I was unable to reproduce the error. I tried both versions of ros2 (the release-beta1, and the current version on master). They both worked fine. 

Some more information, I'm sending a `Uint8MultiArray` of size 2073600. Using the default qos setting for the publisher and subscriber. I'm reading a jpeg image of size (1920x1080) and publishing the jpeg data as integer array. (Python in ros2 cannot handle non-ASCII strings yet, I believe).

wjwwood, 2017-04-08:
Ok, we should still try to get a better error message if we can. But without a reproducible example of it failing that will be hard.

In that case, I'm going to close this as something like "cannot reproduce" and if anyone else runs into it they can comment here and I'll reopen this.

---

P.S. why are you send images as int arrays and not using the `sensor_msgs/Image` type?

closed 2017-04-08 by wjwwood

srsidd, 2017-04-09:
Yes, I will experiment more with the vm to see if something comes up.
Yeah about that, I want to publish raw JPEG data as a string. But python is ros2 does not support non ascii strings, so I had to make it an int array.. I'm not sure if that works though..

wjwwood, 2017-04-10:
If you're sending binary data, you should definitely at least use `uint8[]` and you should not use a `string` type. We have a standard type for compressed images as well:

http://docs.ros.org/api/sensor_msgs/html/msg/CompressedImage.html

That link is for the ROS 1 message, but the same message exists in ROS 2.