ros2/rmw_fastrtps #107: memory leak in topic_name of subscription and publisher (at least)

wjwwood, 2017-05-09:
See:

- `new` called here with no matching `delete`/`free()` that I can find:
  - https://github.com/ros2/rmw_fastrtps/blob/3a7538ae30aa6c883513f0392c024956093dc4ff/rmw_fastrtps_cpp/src/functions.cpp#L879
- `new` called here with no matching `delete`/`free()` that I can find:
  - https://github.com/ros2/rmw_fastrtps/blob/3a7538ae30aa6c883513f0392c024956093dc4ff/rmw_fastrtps_cpp/src/functions.cpp#L1120

Really the whole file should be audited. Also the `new` can throw a C++ exception that would raise through the `rmw` C interface which is bad. It should either use `malloc` or `rmw_allocate` or the new `rcutils_allocator_t` or it should try-catch the `new`.

allenh1, 2017-06-19:
@wjwwood I'll take care of this one.

allenh1, 2017-06-19:
There's also another problem here, in that valgrind usually complains about a mismatched `free()` / `delete` / `delete[]` somewhere in this file (noticed this while working on the python leaking).

wjwwood, 2017-06-19:
There is almost certainly other issues in that file. It wasn't written by us originally, and so hasn't been subject to code review for much of its life. This is issue was about a very particular issue. I don't think you can use valgrind pass/fail as an indicator of whether or not this is fixed.

allenh1, 2017-06-19:
I also see a lot of inconsistency wrt the use of `nullptr`.

mikaelarguedas, 2017-06-19:
Yes as @wjwwood mentioned this file has not been written by us, has a lot of inconsistency and issues and should be both audited and refactored. This is out of the scope of this particular issue as pointed out in the previous comment.

mikaelarguedas, 2017-06-20:
putting this back in ready with a more generic title to reflect the first comment:
> Really the whole file should be audited

mikaelarguedas, 2017-06-20:
closed in favor of https://github.com/ros2/rmw_fastrtps/issues/122

closed 2017-06-20 by mikaelarguedas