ros2/rmw_fastrtps #112: rmw_count_subscribers()/publishers() doesn't take rt topic prefix into account

clalancette, 2017-05-30:
When trying to get an accurate count of publishers or subscribers to a topic by calling `node->count_publishers("topic_name")`, the underlying RMW implementation does not take the topic prefix name into account.  For instance, if you start publishing a topic called "map" in a node by calling:

```
    occupancy_grid_publisher_ =
      node_handle_->create_publisher<::nav_msgs::msg::OccupancyGrid>("map", custom_qos_profile);
```

And then you subscribe to the topic with:

```
$ rostopic_echo_py nav_msgs/OccupancyGrid map
```

And then you try to get a count of subscribers in the original node by calling:

```
int subs = node_handle_->count_subscribers("map");
```

It will always return 0.  Looking at the code a bit, the problem seems to be that down in rmw_fastrtps_cpp/src/functions.cpp:rmw_count_subscribers(), the topic name that comes from the user does *not* have the rt prefix, but the list of topics that is built up expects the rt prefix.  A simple solution is to add the rt prefix to the passed-in string before searching the list.  rmw_count_publishers() has the same problem.

Karsten1987, 2017-06-07:
@clalancette I believe this is fixed. Can you confirm that? So we could close this ticket.

clalancette, 2017-06-08:
Yes, this is fixed now and working as I expect.  I'll close this.  Thanks!

closed 2017-06-08 by clalancette