ros2/rmw_fastrtps #118: Aborted due to: Address already in use

Alex031544, 2017-06-10:
Hello,

until now we are using Opensplice. But with some more than two nodes the messages got transmitted in batches. So we decide to switch over to FastRTPS.

But at the beginning  the program aborted:
```
pi:test$ testprog
running in master mode
starting node testnode
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  set_option: Address already in use
Aborted
```
The programs run on multiple Raspberry Pi 2 with *Rasbian* upgraded to *Stretch*. Our communication is done via an additional 5 GHz WLAN-USB-dongle. On some discussions I recognised that FastRTPS uses only one network interface and chooses for that randomly one of them. On the Raspberry we have three:
```
pi:test$ ifconfig 
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:5d:cb:31  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 2  bytes 98 (98.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 98 (98.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.62  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 fe80::f4ee:1035:89c8:5ce  prefixlen 64  scopeid 0x20<link>
        ether 80:3f:5d:21:f2:ee  txqueuelen 1000  (Ethernet)
        RX packets 16938  bytes 4609241 (4.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1242  bytes 139618 (136.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
A side note: We using ssh to work on the Raspberries using the same WLAN (wlan0). The eth0 is not in use.

How to setup/configure the program correctly to get it running.

Thanks in advance.

allenh1, 2017-06-29:
So I encountered this the other day, and my solution was to source the ros1 ws, remove the currently built ros1_bridge, then rebuild it. After that it just kinda worked. Would be nice if you could give that a go, just to confirm it wasn't some fluke on my end.

allenh1, 2017-07-14:
Can you confirm if you've been able to reproduce this behavior?

allenh1, 2017-07-19:
From [this](https://discourse.ros.org/t/ros2-beta2-on-mac-crashes-with-address-already-in-use/2234/3) post on discourse, it looks like we can close this. Can you confirm this is fixed?

Alex031544, 2017-07-20:
I will look for it soon!

allenh1, 2017-09-06:
@Alex031544 ping

Alex031544, 2017-09-06:
Currently I prepare a setup of virtual machines to investigate this issue.

Alex031544, 2017-09-17:
Hi,
currently I don't get this issue reproduced. Maybe this is due to the newer version.
I got the connection running between three virtual machines. Can you tell me how to measure the traffic with wireshark? I only get visible acknowledges and heartbeats... but not the single packages.

allenh1, 2017-09-19:
> currently I don't get this issue reproduced. Maybe this is due to the newer version.

Great! As such, I'm going to close this. Feel free to reopen if you encounter the problem again.

> Can you tell me how to measure the traffic with wireshark? I only get visible acknowledges and heartbeats... but not the single packages.

I have no idea how to do that. Maybe someone else on the @ros2/team can answer?

closed 2017-09-19 by allenh1

mikaelarguedas, 2017-09-19:
> Feel free to reopen if you encounter the problem again

Feel free to comment here and we will reopen

> Can you tell me how to measure the traffic with wireshark? I only get visible acknowledges and heartbeats... but not the single packages.

Metadata packets are sent on all interfaces while the data is sent only on the selected one (e.g. if I run nodes on a single machine data will go on the loopback interface while metadata will also go on my wired/wifi interface). Haven't tried monitor traffic between VMs though, maybe listening to the interface doing the bridge between VMs will allow you to see data packets