ros2/rmw_fastrtps #125: Best effort publisher stops sending to local subscribers when remotes connect

dhood, 2017-06-23:
**Issue:**

With the 1.5.0 fastrtps release the following issue has appeared:

1. best effort publisher talking to best effort subscriber(s) on the same machine.
1. best effort subscriber connects on the same domain ID but a remote machine.
1. best effort subscriber connected remotely receives message, but the local subscriber(s) does not.
1. when the remote subscriber is killed, the local subscribers receive messages again.
1. repeat. whenever the remote subscriber is connected, the local ones do not receive messages.

**Steps to reproduce:**
Machine A:
```
talker_qos_py -n 100
listener_qos_py -n 100
```
These use `/chatter_qos` topic and **best effort** publisher and subscriber. They talk fine.

Machine B:
```
ROS_DOMAIN_ID=<machine_A_domain_ID> listener_qos_py -n 100
```
Receives messages and causes listeners on machine A to stop receiving messages.

---

**Workaround:**

The issue does not happen if the publisher offers reliable (even if the subscriber and therefore connection is best effort).

Example:

Machine A:
```
talker -t chatter_qos
listener_qos_py -n 100
```
This starts a publisher that **offers reliable** reliability publishing on the `chatter_qos` topic. (they should still communicate with best effort).

Machine B:
```
ROS_DOMAIN_ID=<machine_A_domain_ID> listener_qos_py -n 100
```

---

**Ideas:**

@wjwwood and I were brainstorming and wonder if there have been changes to the unicast/multicast settings of publishers. maybe the best effort publisher switches between unicast and multicast when subscribers are present both locally and remote, but there's a but that gets the loopback cut off in the process.

I have not yet tried to reproduce this with ros-agnostic fastrtps demos. As there's a workaround, this isn't necessarily a blocker for beta2 demos provided we switch publishers to reliable.

dhood, 2017-06-23:
The practical impact of this on beta2 is that remote introspection of topics on the turtlebot (eg echo, or visualising with rviz via the bridge) causes nodes on the turtlebot to stop communicating (without the workaround)

clalancette, 2017-06-23:
@dhood Does this happen immediately for you, or is it a slow decline in performance?  With the cartographer demo (where I have almost everything best-effort, including the map topic), I don't see this.  I am able to run the dynamic_bridge remotely, and have it subscribe to topics, and keep up.  That being said, I've only ever tested it for up to 10 minutes at a time, so if it is something that happens over time, I wouldn't have noticed it.  I'm just trying to figure out what is different in our setups.

dhood, 2017-06-23:
it's instant. in the case of AMCL, it's the laser scan that stops being received whenever someone is introspecting from another machine (including @wjwwood's). the only reason I noticed it with amcl is because there's a timer that reports if laser messages aren't being received. with cartographer you might not notice that the scans stop being received. I imagine if you start the remote subscriber before starting carto, carto will never receive any scans (provided you aren't using the workaround in https://github.com/ros2/depthimage_to_laserscan/issues/2)

clalancette, 2017-06-23:
Oh, I see.  That's an interesting problem, I don't think I've specifically looked at that.  The /map topic is published by cartographer itself, so wouldn't be subject to the problem, and I believe the /tf topic (the only other one that I send off-robot) is reliable, so also doesn't see this.  So that explains why I don't see the problem.

dhood, 2017-06-23:
I would expect you to see the equivalent thing with cartographer when subscribing to one of the other topics being adverstised with best_effort, e.g. [currently point cloud or map](https://github.com/ros2/cartographer_ros/blob/6cd3bffc51d924e97e5f08bf7500e33a5e7bc26f/cartographer_ros/cartographer_ros/node.cc#L133). rviz might be using service calls instead of topic subscriptions for the submaps

dhood, 2017-06-28:
Addressed upstream

closed 2017-06-28 by dhood