ros2/rmw_fastrtps #127: Bad rmw_count_subscribers

fmrico, 2017-06-27:
I am using the master branch with Fast-RTPS.

I've been testing the ros1_bridge with one ros1 talker and two ros2 listener. 

[image]

When running, I checked on introspection that it only showed that there was 1 subscriber in /chatter, instead of the two that were active. 

[image]

When I interrupted one of the ros2 listener, the other stopped receiving, and the instrocpection shows that there is no subscribers for that topic.

[image]

I have debugged a bit and I have reached rmw_fastrtps/rmw_functions.cpp, where it seems that the rmw_count_subscribers function does not return the correct value.

allenh1, 2017-06-27:
Thanks for the report! After playing around with this on my machine, I was not able to reproduce this exact behavior, but I did find something a little off:

```
allenh1@machina:~/ros2_ws$ ros2 run ros1_bridge dynamic_bridge
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
...
[ERROR] [1498601222.019055259]: getifaddr() failed
Couldn't find an AF_INET address for [machina.local]
Failed to look up /talker/set_logger_level
Couldn't find an AF_INET address for [machina.local]
Failed to look up /rosout/get_loggers
Couldn't find an AF_INET address for [machina.local]
Failed to look up /talker/get_loggers
Couldn't find an AF_INET address for [machina.local]
Failed to look up /rosout/set_logger_level
Couldn't find an AF_INET address for [machina.local]
Failed to look up /ros_bridge/get_loggers
Couldn't find an AF_INET address for [machina.local]
[ERROR] [1498601222.019398046]: [getPublishedTopics] Failed to contact master at [machina.local:11311].  Retrying...
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
Couldn't find an AF_INET address for [machina.local]
```

(there were many "passing message from ROS 1 to ROS 2"'s that I removed to clean up the output).

But, if I simply reran the dynamic_bridge, everything was fine. I was even able to have a ros1 cpp listener on my laptop, two ros 2 cpp listeners on my desktop, and a ros2 python listener on my desktop. 

---

After further testing, I was able to reproduce the above behavior without killing anything. The bridge just dies ~1100 messages in. I'm going to test this with connext to see if this is a fastrtps problem.

allenh1, 2017-06-27:
Ok, I can now confirm that this both affects fastRTPS and connext.

allenh1, 2017-06-27:
@fmrico How recent is your master?

fmrico, 2017-06-28:
@allenh1 My master was updated yerterday. I have updated right now and same behavior. When you are running two ROS2 listeners, how many ROS2 subs are showed in the ros1_bridge introspection messages? I am still seeing 

`ROS 2: /chatter (std_msgs/String) [0 pubs, 1 subs]`

I think that I should see `2 subs`. 

I also experienced the error you comment on getifaddr() ros2/ros1_bridge#78. In fact, there seems to be a relationship between the number of listeners and the number of messages to which the bridge stopped working. With two listeners the bridge died ~ 600 messages.

I thought it was not related to my problem.

dirk-thomas, 2017-06-28:
Can you please retry with the latest master with #79 being merged. Or was that spray the state you used when you mentioned:

> After your last push

fmrico, 2017-06-28:
@dirk-thomas After applying your commit cf5108e in dynamic_bridge.cpp the problem ros2/ros1_bridge#78 and ros2/ros1_bridge#79 seems to be **fixed** (I waited until msg number ~5000 running two ROS2 listeners) both in Fast-RTPS and RTI Connext. 

Fast-RTPS keeps failing counting subscribers...

allenh1, 2017-06-28:
@fmrico I think that the issue there has to do with the fact that the subscribers have the same name -- we consider such a case to be undefined behavior. In the future, we might consider this case, but, as of yet, we have no plans to do so.

You can check out the design doc [here](http://design.ros2.org/articles/changes.html).

dirk-thomas, 2017-06-28:
> I think that the issue there has to do with the fact that the subscribers have the same name -- we consider such a case to be undefined behavior.

The `ros1_bridge` doesn't consider node names at all (only for preventing to loop messages between publishers and subscribers the bridge created itself). Therefore the node names on the ROS 1 and ROS 2 side can't be the reason.

allenh1, 2017-06-28:
@fmrico I'm able to reproduce that behavior now. I have two C++ subscribers and a Python subscriber, yet the messages show only one subscriber. I'm looking into this right now.

allenh1, 2017-06-28:
The bridge pulls the subscriber value from rclcpp, which calls the `count_publishers` and `count_subscribers` functions from `node_graph.cpp`, which calls `rmw_count_publishers`. I'm going to see if the counts are incorrect for connext as well.

allenh1, 2017-06-28:
Ok, cool.

```
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  Passing message from ROS 1 to ROS 2
  ROS 2: /chatter (std_msgs/String) [0 pubs, 3 subs]
  ROS 2: /parameter_events (rcl_interfaces/ParameterEvent) [3 pubs, 0 subs]
```

this appears to only affect fastRTPS. I'll try and find a fix for it.

allenh1, 2017-06-28:
8d10e43 fixed the count up, now I just need to fix the count down.

dirk-thomas, 2017-06-28:
This should be fixed by #129.

closed 2017-06-28 by dirk-thomas