ros2/rmw_fastrtps #131: refactor functions.cpp

Karsten1987, 2017-07-03:
This is a top level issue for the refactoring task. The idea is to refactor the `_functions.cpp` into multiple files, making the package easier to maintain.

dirk-thomas, 2017-07-03:
Please hold off to start any work on this until we can sync and ensure that only one person touches the repo during that refactoring.

Karsten1987, 2017-07-04:
yes agreed. I'll plan on getting rmw_fastrtps_cpp as well as rmw_connext_cpp refactored within one week. What is the best way to sync on this one?

dirk-thomas, 2017-07-07:
We talked about this briefly in yesterdays meeting. Nobody else will touch the two repos `rmw_connext` and `rmw_fastrtps` for the time being. So you have exclusive access and can refactor both in a similar way as `rmw_opensplice_cpp`. Please make sure to structure the patch like this:

* First commit copying the `functions.cpp` to different file names (`.cpp` as well as `.h` / `.hpp`) without any changes. This will make sure history is browsable.
* Second commit removing blocks of code from all new files not needed in them (as well as the obsolete `functions.cpp`). This will make sure that its easy to review.
* Third commit update CMake, includes etc. to make the code compile again.
* Any additional changes you would like to apply. Keeping those separate from the "code shuffling" is important to make the review easier.

dirk-thomas, 2017-07-14:
Addressed by #135.

closed 2017-07-14 by dirk-thomas