ros2/rmw_fastrtps #137: Sending string results into eprosima::fastcdr::exception::NotEnoughMemoryException

firesurfer, 2017-07-28:
When sending this string:
"No environment file found at /etc/systemd/system/iboss_environment, creating own
QIODevice::write (QFile, "/etc/systemd/system/iboss_environment"): device not open"

I run at the recieving node into this exeception:
```
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
Abgebrochen

```

This doesnt happen if I use the ros2 tool:
```
 ros2 topic  pub /ros2_log ros2imple_logger/LoggingMessage "{\"level\":3, \"nodename\":\"Scheduling_Node100\", \"message\":\"No environment file found at /etc/systemd/system/iboss_environment, creating own\nQIODevice::write (QFile, \\\"/etc/systemd/system/iboss_environment\\\"): device not open\"}"

```

After further research I found that using `rmw_qos_profile_sensor_data` was the cause of this trouble. Using the default profile (what the ros2 tool probably does doesn't invoke this error"

mikaelarguedas, 2017-08-02:
Thanks @firesurfer for reporting this. Can you provide a [small reproducible example](http://sscce.org/) for this?

firesurfer, 2017-08-03:
Seems like a recent update of fast-rtps or of ros2 fixed this bug. Closing it!

closed 2017-08-03 by firesurfer