ros2/rmw_fastrtps #139: Update time for get_node_names can be very low

firesurfer, 2017-08-07:
See the discussion on discourse: https://discourse.ros.org/t/update-time-for-get-node-names/2333

The update time of the `get_node_names` function is very low in case a node was shutdown (Around 5 minutes). In case a node was started the update is almost immediately. 

@Karsten1987 
@richiprosima 

I can't provide the real code where we found this behavior but I will provide a simple example as soon as possible.

dhood, 2017-09-15:
@firesurfer we have changed the way that sigints are handled by `ros2 run` in https://github.com/ros2/ros2cli/pull/50 so that nodes are able to shutdown cleaner. Are you able to update to beta3/master and let us know if you still see this issue in your context?

firesurfer, 2017-10-16:
Yes we still see this issue. Is it necessary to run nodes via `ros2 run` in order for a proper shutdown?

dirk-thomas, 2017-10-20:
No, `ros2 run` is just a different way to invoke the executable.

I will mark this with "more-information-needed" until there is a simple example to reproduce the problem.

clalancette, 2023-06-13:
I'm going to close this out, since things have changed significantly since this time.  Feel free to open a new issue if you are still having problems.

closed 2023-06-13 by clalancette