ros2/rmw_fastrtps #149: exported libraries should not expose imported targets of FastRTPS / FastCDR

dirk-thomas, 2017-08-24:
See #148 for the current workaround to make this package usable by downstream packages.

sloretz, 2018-04-23:
If I understand correctly then the workaround in #148 seems reasonable to me. Is the issue that `rmw_fastrtps_cpp_LIBRARIES` contains an IMPORTED target from an upstream library? Would it be fine if it contained an IMPORTED target from the current library, or does it need to have full paths like [recommended for xxx_LIBRARIES in find-modules](https://cmake.org/cmake/help/v3.5/manual/cmake-developer.7.html#standard-variable-names)?

I think the workaround is a symptom of defining variables instead of following [cmake's recommendation that packages export their targets](https://cmake.org/cmake/help/v3.5/manual/cmake-packages.7.html#creating-relocatable-packages). Since the target `rmw_fastrtps_cpp` has transitive dependency info but `ament_cmake` is not exporting it, it seems appropriate to fix this in `ament_export_libraries()`.

Option 1 is closer to cmake's recommendation and preserves other transitive target information like compiler options and include directories. Option 2 seems more aligned with what I assume was an intentional design choice to make `ament_cmake` only provide variables. Thoughts or other strategies? I'll make a PR for Option 1 tomorrow if I hear nothing.

## Option 1
### export target `rmw_fastrtps_cpp`, install exported target file, and append the target name to `rmw_fastrtps_cpp_LIBRARIES`

The variable `rmw_fastrtps_cpp_LIBRARIES` would contain an imported target `rmw_fastrtps_cpp::rmw_fastrtps_cpp`.

The macro `ament_export_libraries()` would call `export()` and `install(EXPORT)` for each argument that is a non-IMPORTED target. The exported target name would be stored in `_AMENT_EXPORT_LIBRARIES` and the exported file name in a new variable `_AMENT_EXPORT_TARGET_FILES`. The template `ament_cmake_export_libraries.cmake.in` would need to `include()` these exported target files.

## Option 2
### Iterate through `INTERFACE_LINK_LIBRARIES` of targets starting at `rmw_fastrtps_cpp`, and add the library location to `rmw_fastrtps_cpp_LIBRARIES`

The variable `rmw_fastrtps_cpp_LIBRARIES` would contain paths to libraries.

`ament_cmake_export_libraries_package_hook.cmake` would iterate through the content of `_AMENT_EXPORT_LIBRARIES`. If an item is a target then it would iterate through its `INTERFACE_LINK_LIBRARIES`. If any item in there is a path then it gets added to `_AMENT_EXPORT_LIBRARIES`, else if it is a target then the routine recursively checks `INTERFACE_LINK_LIBRARIES` of the target.

dirk-thomas, 2018-04-27:
I think option 1 is the desired one. The package would "just" need to use the functionality provided by the [ament_cmake_export_interfaces](https://github.com/ament/ament_cmake/tree/f74ccfa7399156d20c8d0eff8100445b7eefebd9/ament_cmake_export_interfaces) package.

Unfortunately until now only a single ROS package makes use of that: [ament_index_cpp](https://github.com/ament/ament_index/blob/d9f7f2783dedc5687a5bdaf3ce5252dce5c073c5/ament_index_cpp/CMakeLists.txt#L37)

sloretz, 2018-05-03:
Closing since #198 is merged

closed 2018-05-03 by sloretz