ros2/rmw_fastrtps #150: Allow configuration of the severity level for eProsima logging

deng02, 2017-08-25:
Feature Request

Currently the eProsima logging level is hardcoded to ERROR:
https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/src/rmw_node.cpp#L66

For troubleshooting, it would be useful to be able to change the severity level of the eProsima to INFO or DEBUG, either independently of ROS2 or in conjunction with.

dirk-thomas, 2017-08-25:
I think the best path forward is to implement a custom `LogConsumer` for FastRTPS which uses the ROS logging API provided by `rcutils`. Currently `rcutils` doesn't offer name-specific severity levels but that feature is planned to be added in the future.

If you would like to see this feature rather earlier than later we would appreciate a PR for it.

MiguelCompany, 2018-08-29:
I think this can be done calling [rmw_set_log_severity](https://github.com/ros2/rmw_fastrtps/blob/7bb28917586988299e7ccdfd0c209b259c9e8fbc/rmw_fastrtps_cpp/src/rmw_logging.cpp#L25) which in turn calls [this](https://github.com/ros2/rmw_fastrtps/blob/7bb28917586988299e7ccdfd0c209b259c9e8fbc/rmw_fastrtps_shared_cpp/src/rmw_logging.cpp#L29)

dirk-thomas, 2018-08-29:
Implemented a while ago in #156.

closed 2018-08-29 by dirk-thomas