ros2/rmw_fastrtps #159: Weird cmake dependencies on  rosidl_typesupport_introspection_c, rosidl_typesupport_introspection_cpp

jwang11, 2017-10-19:
In rmw_fastrtps_cpp/CMakeLists.txt, I see some weird dependencies as below. I don't see the reason that rmw implementation depend on rosidl_xxx packages. if or not I misunderstand sth? 
ament_target_dependencies(rmw_fastrtps_cpp
  "rcutils"
  "rosidl_typesupport_introspection_c"
  "rosidl_typesupport_introspection_cpp"
  "rmw"
  "rosidl_generator_c"
  "rosidl_generator_cpp")

dhood, 2017-10-19:
ROSIDL is the message format of ROS messages, and the rosidl_* packages contain common code for working with those messages that is also used by other rmw implementations. For example the typesupport_introspection packages are also used by `rmw_connext_dynamic_cpp` as it's another rmw implementation that supports runtime interpretation of ROS msgs via "introspection" using msg metadata, as opposed to relying on pre-built support for each msg type.

Have a read of http://docs.ros2.org/beta3/developer_overview.html#internal-ros-interfaces which goes into more detail about typesupport and the roles the packages, poke around the codebase to see how the packages relate to each other, and feel free to post back with a more specific question if something's still unclear

jwang11, 2017-10-19:
Thank you for explanation. I basically understand what rosidl_XXX do, however, rmw_fastrpts not use .msg or .svr, thus no need to generate c/cpp code for them.  I tried to drop those rosidl_xxx, there is no compile issue.

dhood, 2017-10-19:
Things from `rosidl_generator_c` are used during serialisation/deserialisation: https://github.com/ros2/rmw_fastrtps/blob/a6ac02905c55118fb1da636f85c8197dde8ef58e/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/TypeSupport.hpp#L85
so if you got it to build you maybe weren't using a clean, isolated workspace.

For rosidl_generator_cpp I'm not sure where the dependency comes from, it might be safe to remove it

jwang11, 2017-10-19:
I searched the code, rosidl_generator_cpp did not used, however rosidl_typesupport_introspection_cpp was widely used. Is it the normal case?

dhood, 2017-10-20:
It depends. rmw_fastrtps_cpp needs to know about the c data type that's being used for strings (what I linked to before) which is why it depends on rosidl_generator_c. For c++, it doesn't need to depend on rosidl_generator_cpp to get any info about the string data type because it's just an std::string (if it were a custom type then it would).

as far as I know there isn't any reason for the rosidl_generator_cpp dependency. I proposed its removal in https://github.com/ros2/rmw_fastrtps/pull/161

mikaelarguedas, 2017-10-20:
> I tried to drop those rosidl_xxx, there is no compile issue.

Quick follow up on this. The reason it succeeds to build when removing rosidl_generator_c is because its headers are exported by [rosidl_typesupport_introspection_c](https://github.com/ros2/rosidl/blob/52830d1e45ad3f3cb470dfd3fd093735a205ead1/rosidl_typesupport_introspection_c/CMakeLists.txt#L18).
Though the dependency on rosidl_generator_c should be kept because we use it's headers and symbols directly in this package as pointed out by @dhood

jwang11, 2017-10-20:
That is fair. thanks

closed 2017-10-20 by dhood