ros2/rmw_fastrtps #179: rmw_fastrtps does not detect the version correctly when checking RMW_IMPLEMENTATION

tfoote, 2017-12-06:
Testing https://github.com/ros2/ros2/wiki/Working-with-multiple-RMW-implementations#ros-2-ardent-and-later

Against the binary build on windows: http://ci.ros2.org/view/packaging/job/packaging_windows/825/

It fails to error correctly when evaluating an invalid RMW_IMPLEMENTATION however the opensplice build works as expected.

```
>ros2 run demo_nodes_cpp talker
[ERROR] [rcl]: Error getting RMW implementation identifier.
```

Fuller console outputs  https://gist.github.com/tfoote/1533d84fa2668127601313ab91883735

However it appears to pass in CI with only fastrtps: http://ci.ros2.org/job/ci_windows/3845/testReport/junit/rcl.build.rcl.test.test/test_rmw_impl_id_check__rmw_fastrtps_cpp_Release/test_rmw_implementation_env/

dirk-thomas, 2017-12-06:
I tried the multi-rmw fat archive on Linux (from [packaging_linux#898](http://ci.ros2.org/view/packaging/job/packaging_linux/898/)) and it works correctly for switching between FastRTPS and OpenSplice.

dirk-thomas, 2017-12-06:
In the single-rmw case the API will return the identifier of the rmw implementation chosen at compile/link time. So the answer will be something like:

> you wanted 'foo' but got 'bar'

In the multi-rmw case the API will use the hint from the env var to load the desired rmw impl. and will fail do to so and answer something like:

> can't get rmw impl identifier

So I think you are comparing two different things here.

tfoote, 2017-12-07:
Resolved in: https://github.com/ros2/rcl/pull/201

closed 2017-12-07 by tfoote