ros2/rmw_fastrtps #181: Non-ascii character in package.xml

racko, 2017-12-18:
The "offending" character is actually in the maintainer's name, so I'm not sure how to deal with this:

https://github.com/ros2/rmw_fastrtps/blob/640d7de7e2af1a4281ba910d0ed6702e337ed48c/rmw_fastrtps_cpp/package.xml#L7

```
+++ Building 'rmw_fastrtps_cpp'
Running cmake because arguments have changed.
==> '. /home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/sbin/cmake /home/racko/src/ros2-1.0.0/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/racko/src/ros2-1.0.0/install' in '/home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp'
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /usr/sbin/cc
-- Check for working C compiler: /usr/sbin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/sbin/c++
-- Check for working CXX compiler: /usr/sbin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake_ros: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/ament_cmake_ros/cmake)
-- Found PythonInterp: /usr/sbin/python3 (found suitable version "3.6.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/sbin/python3
-- Found rcutils: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /home/racko/src/ros2-1.0.0/install/include  
-- Found rmw: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/rmw/cmake)
-- Found rosidl_typesupport_introspection_c: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/rosidl_typesupport_introspection_c/cmake)
-- Found rosidl_typesupport_introspection_cpp: 0.4.0 (/home/racko/src/ros2-1.0.0/install/share/rosidl_typesupport_introspection_cpp/cmake)
Error parsing '/home/racko/src/ros2-1.0.0/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml':
Traceback (most recent call last):
  File "/home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 145, in <module>
    main()
  File "/home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 56, in main
    raise e
  File "/home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 53, in main
    package = parse_package_string(args.package_xml.read())
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 409: ordinal not in range(128)
CMake Error at /home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/usr/sbin/python3
  /home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/racko/src/ros2-1.0.0/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml
  /home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /home/racko/src/ros2-1.0.0/install/share/ament_cmake_core/cmake/core/ament_package.cmake:54 (ament_package_xml)
  CMakeLists.txt:116 (ament_package)

-- Configuring incomplete, errors occurred!
See also "/home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp/CMakeFiles/CMakeOutput.log".
<== Command '. /home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/sbin/cmake /home/racko/src/ros2-1.0.0/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/racko/src/ros2-1.0.0/install' failed in '/home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp' with exit code '1'

<== Command '. /home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/sbin/cmake /home/racko/src/ros2-1.0.0/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp -DBUILD_TESTING=0 -DCMAKE_INSTALL_PREFIX=/home/racko/src/ros2-1.0.0/install' failed in '/home/racko/src/ros2-1.0.0/build/rmw_fastrtps_cpp' with exit code '1'
==> ERROR: A failure occurred in build().
    Aborting...
```

mikaelarguedas, 2017-12-18:
Did you try setting your locale to UTF-8 ?

(Snippet copied from https://github.com/ros2/ros2/wiki/Linux-Development-Setup):
```
sudo locale-gen en_US en_US.UTF-8
sudo update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
export LANG=en_US.UTF-8
```

racko, 2017-12-18:
aaaah! I'm currently setting up an Arch Linux docker container to be able to reproducibly build a ros2 package. I forgot to follow this step of the guide. And I was wondering why that error didn't occur when I built it on my machine ... thx!

closed 2017-12-18 by racko

mikaelarguedas, 2017-12-18:
> I'm currently setting up an Arch Linux docker container ...

Something I faced in the past that may save you a headache:
Since last year most base images don't ship with `locales` so you'll have to install the package in your Dockerfile.
You'll also need to set up the locale in your Dockerfile as changing it within a container has no effect.
See our [CI dockerfile](https://github.com/ros2/ci/blob/35b86c8492ea5e80c0653bedfa403d3ab4717acc/linux_docker_resources/Dockerfile#L10-L12) for an example.

Another solution that should avoid aving to install anything could be (though I havent tried it):
```
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8
```