ros2/rmw_fastrtps #184: rmw_fastrtps CustomParticipantInfo structure defination.

sagniknitr, 2018-01-02:
`typedef struct CustomParticipantInfo`
`{`
`eprosima::fastrtps::Participant * participant;`
`ReaderInfo * secondarySubListener;`
`WriterInfo * secondaryPubListener;`
`rmw_guard_condition_t * graph_guard_condition;`
`} CustomParticipantInfo;`

What is the purpose of the secondarySubListener in the above strucutre, which is implemented in custom_participant_info.hpp ?

dhood, 2018-01-02:
Each node has `secondarySubListener` and `secondaryPubListener` to get information about subscribers and publishers in the ROS graph (respectively). These get attached to the node [when it's created](https://github.com/ros2/rmw_fastrtps/blob/97a099836235f9b56bea16c5d1784e103ffdf777/rmw_fastrtps_cpp/src/rmw_node.cpp#L140) and process graph events in the background. The information can be queried by functions such as [`rmw_get_topic_names_and_types`](https://github.com/ros2/rmw_fastrtps/blob/3e5edb3316e2aee325e9354549e54c999a89bca9/rmw_fastrtps_cpp/src/rmw_topic_names_and_types.cpp#L74). Another way they're used is that they can [trigger a guard condition](https://github.com/ros2/rmw_fastrtps/blob/3742a118070e5e2467fe1888fc586f6ea49f8b5e/rmw_fastrtps_cpp/include/rmw_fastrtps_cpp/reader_info.hpp#L102) when graph changes occur, which will wake up `rmw_wait` in the event that something like `wait_for_service` is being called (waiting for the topic corresponding to that service to become available).

sagniknitr, 2018-01-03:
Thanks for the clarification.Do they also use the DDS for sending/receiving query information to/from other nodes ?

sagniknitr, 2018-01-03:
@dhood Thanks for the clarification.Do they also use the DDS for sending/receiving query information to/from other nodes ?

dhood, 2018-01-08:
The `rmw` abstraction layer itself is not specifically tied to DDS, but in the specific case of the rmw_fastrtps implementation of the rmw layer, this discovery information is exchanged using DDS/RTPS, yes.

closed 2018-01-17 by sagniknitr