ros2/rmw_fastrtps #188: ICMP - Destination unreachable ( Port unreachable)

firesurfer, 2018-02-05:
Hi as mentioned in #157 I could track down the reason for the ICMP - Destatination unreachable message one can obtain using wireshark.

How to reproduce:

1. Start the example publisher on the first computer
2. Start the example listener on the second computer (they should transmit messages between each other)
3. Start wireshark
3.1 you should now see RTPS data and Heartbeats
4. Kill  the listener (not SIGTERM but realy SIGKILL -> this simulates a crash)
5. You should now see the remaining application / computer still send RTPS Heartbeats that then get answered by `ICMP - Destination unreachable ( Port unreachable)`

Tested on a laptop running Debian Testing and a Raspberry Pi3 running the latest ros2 master branches. 
We could reproduce this error also an a computer running ubuntu 16.04

mikaelarguedas, 2018-02-08:
Hi @firesurfer,

This looks more like a protocol behavior than a ros2 specific behavior to me. My understanding of the DDS spec is that if a participant is killed without being given a change to notify the others that it's going down, the other ones will assume it'll eventually get back "in range" in a given (configurable) amount of time before considering it dead (at which point it will stop sending heartbeats).

Could you provide more details about what behavior you are expecting to see in the scenario you described ? 
Thanks!

firesurfer, 2018-02-09:
Hi @mikaelarguedas 

I actually expect this behaviour from the protocol but there are tree things:

1. What amount of time is configured in ROS2 (fast rtps) and how could I change it
2. This behaviour also exists if the application is terminated via SIGTERM and not SIGINT (this is what systemd does by default)
3. On all of our debian configurations the nodes (that run fine on ubuntu) seam to exit like they are killed (Also I need to press STRG + C twice). I guess this is due to other compiler + library version. (I guess this isn't your problem because only ubuntu is supported by ROS2)

In total 2 and 3 make our configuration spam a lot unused heartbeats over the network. 
I would expect that the exit handler implemented in ROS2 also handles SIGTERM and not only SIGINT.

closed 2023-07-17 by firesurfer