ros2/rmw_fastrtps #191: ROS2 subscriber crashes with NotEnoughMemoryException with empty effort array in JointState messages

jespersmith, 2018-03-17:
## Bug report

**Required Info:**

- Operating System:
 - Ubuntu 16.04 64 bit
- Installation type:
  - Binary debian packages from deb http://repo.ros2.org/ubuntu/main xenial main
- Version or commit hash:
  - deb http://repo.ros2.org/ubuntu/main xenial main
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - rclpy with command line

#### Steps to reproduce issue

```
ros2 topic pub /eve/joint_states sensor_msgs/JointState "{name: [a,a,a,a,a,a,a], position: [128.0,128.0,128.0,128.0,128.0,128.0,128.0], velocity: [128.0,8.0,128.0,128.0,128.0,128.0,128.0]}"

ros2 topic echo /eve/joint_states sensor_msgs/JointState
```

#### Expected behavior
The ros2 topic echo process doesn't crash

#### Actual behavior
ros2 topic echo crashes with the following message
```
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
Aborted (core dumped)
```
#### Additional information

Only crashes if the number of elements in the JointState.name, JointState.position and JointState.velocity >=7 and the number of elements in JointState.effort = 0. It seems to work if there is at least one element in effort.

richiware, 2018-03-19:
I was able to reproduce the crash. I've updated Fast-CDR to solve this issue. It was a problem with alignment and zero length arrays at the end of the buffer. Commit eProsima/Fast-CDR@581bc96 contains the solution.

closed 2018-03-19 by jespersmith

richiware, 2025-05-05:
Unfortunately, we currently don't have the resources to address this. However, please feel free to email support@eProsima.com so we can further find a solution for you.