ros2/rmw_fastrtps #193: After update of ros2 workspace -> introduces build error - UnicodeDecodeError

firesurfer, 2018-04-01:
After updating the ros2 workspace (I'm going with the master branches) the rmw_fastrtps_cpp package fails to build with the error:

  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 409: ordinal not in range(128)

Operatingsystem: Debian testing
Compiler: clang 5.0.1
Buildsystem: Ninja
Python Version: Python 3.6.5rc1

It also happens when using GCC 7.3 and make

According to stackoverflow that might be a unicode problem with python, but I couldn't figure out where exactly to fix it in ament: 
https://stackoverflow.com/questions/18649512/unicodedecodeerror-ascii-codec-cant-decode-byte-0xe2-in-position-13-ordinal

The complete build log for that package is:
--------------------------------------------------------------------------------
+++ Building 'rmw_fastrtps_cpp'
==> '. /home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_CXX_FLAGS=-fuse-ld=gold -DCMAKE_BUILD_TYPE=Debug -G Ninja -DCMAKE_INSTALL_PREFIX=/home/firesurfer/workspace/ros2_ws/install /home/firesurfer/workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp' in '/home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp'
-- Found ament_cmake_ros: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_ros/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rcutils: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found rmw: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/rmw/cmake)
-- Found rosidl_typesupport_introspection_c: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/rosidl_typesupport_introspection_c/cmake)
-- Found rosidl_typesupport_introspection_cpp: 0.4.0 (/home/firesurfer/workspace/ros2_ws/install/share/rosidl_typesupport_introspection_cpp/cmake)
Error parsing '/home/firesurfer/workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml':
Traceback (most recent call last):
  File "/home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 145, in <module>
    main()
  File "/home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 56, in main
    raise e
  File "/home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 53, in main
    package = parse_package_string(args.package_xml.read())
  File "/usr/lib/python3.6/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 409: ordinal not in range(128)
CMake Error at /home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/usr/bin/python3
  /home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/firesurfer/workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml
  /home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /home/firesurfer/workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package.cmake:63 (ament_package_xml)
  CMakeLists.txt:119 (ament_package)

-- Configuring incomplete, errors occurred!
See also "/home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp/CMakeFiles/CMakeOutput.log".

<== Command '. /home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_CXX_FLAGS=-fuse-ld=gold -DCMAKE_BUILD_TYPE=Debug -G Ninja -DCMAKE_INSTALL_PREFIX=/home/firesurfer/workspace/ros2_ws/install /home/firesurfer/workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp' failed in '/home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp' with exit code '1'
<== Command '. /home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/cmake -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DCMAKE_CXX_FLAGS=-fuse-ld=gold -DCMAKE_BUILD_TYPE=Debug -G Ninja -DCMAKE_INSTALL_PREFIX=/home/firesurfer/workspace/ros2_ws/install /home/firesurfer/workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp' failed in '/home/firesurfer/workspace/ros2_ws/build/rmw_fastrtps_cpp' with exit code '1'

dirk-thomas, 2018-04-01:
Please check your locale and ensure that it is set to utf-8.

firesurfer, 2018-04-01:
Thanks for your fast answer. My locales are set to:

de_DE.UTF-8 UTF-8
and
en_US.UTF-8 UTF-8

I had to enable the en_US.UTF-8 UTF-8 due to a problem with another program. Disabeling it and regnerating the locales seems to work. 

A valid workaround (in case multiple locales are needed) is to explicitly set the locale before starting the build:
`LC_ALL=de_DE.UTF-8`

or `en_US.UTF-8` 

Thanks for your help! 
I'm still not sure why python has problems with multiple locales enabled.

firesurfer, 2018-04-05:
That is quite simply:
edit `etc/locale.gen` to enable locales for example for german utf8 and english utf8
run `sudo locale-gen` to regenerate the locales.
Make sure to close and reopen your terminal (so the new locales got loaded) and build a ros2 workspace. (Perhaps delete build and install folder before)

mikaelarguedas, 2018-04-05:
I am asking as I was unable to reproduce the issue.
You didn't have to change any of the locales related environment variables ?

Even if I have several locales enabled in `etc/locale.gen` (French, English and German all UTF8), the build works and `LC_ALL` and `LANG` always refer to only one of the locales.
It would be helpful to know the value of your `LANG` and `LC_ALL` variables when it fails

firesurfer, 2018-04-05:
`LANG=de_DE.UTF-8`
the LC_ALL variable isn't set by default. If I set it by hand to `LC_ALL=de_DE.UTF-8` the build works fine. 

According to https://wiki.debian.org/Locale the LC_ALL variable shouldn't be set by default.

richiware, 2018-04-06:
I have had same problem in some systems. It is due to my surname in this [line](https://github.com/ros2/rmw_fastrtps/blob/85568b20419db8aec54dd8bdeb375a1b31332c94/rmw_fastrtps_cpp/package.xml#L7), because it contains an accent. Maybe it is better to remove it.

mikaelarguedas, 2018-04-09:
> Maybe it is better to remove it.

While this would be a temporary workaround, many ROS users have unicode characters in their names so asking them to change their name in the package.xml is not a desired outcome.

> `LANG=de_DE.UTF-8`
the LC_ALL variable isn't set by default. If I set it by hand to LC_ALL=de_DE.UTF-8 the build works fine.

Thanks @firesurfer for the additional information. Mine was set as per the [ROS 2 installation instructions](https://github.com/ros2/ros2/wiki/Linux-Development-Setup), I guess that's why I couldnt reproduce. In the meantime it looks like this was a known problem that motivated adding the override to the installation instructions.

Unsetting it locally doesn't allow me to get the build to fail either. I'll try in a fresh environment with `LC_ALL` not set (not sure when I'll get around it though). If you happen to have a fix or time to investigate I'm more than happy to review / test it as it may allow a quicker resolution of this issue.

firesurfer, 2018-04-17:
@mikaelarguedas 
I didn't followed this part of the installation instruction because it worked fine for me before. After I read the according debian wiki page I don't think it is a good idea to force setting LC_ALL. 

Unfortunatly I not familiar enough with python to provide a fix. My first attempt to google that issue came up with two solutions. Either force a locale inside the python script or tell python to explicitly interpret a string in utf-8. (See: https://stackoverflow.com/questions/18649512/unicodedecodeerror-ascii-codec-cant-decode-byte-0xe2-in-position-13-ordinal)

In our case it isn't this trivial I think because ament is using a xml parser library. Perhaps it is possible to tell the library to use utf8 explicitly.

mikaelarguedas, 2018-04-17:
> After I read the according debian wiki page I don't think it is a good idea to force setting LC_ALL.

Agreed

> In our case it isn't this trivial I think because ament is using a xml parser library. Perhaps it is possible to tell the library to use utf8 explicitly.

Yes that is my guess as well.

I unfortunately haven't been able to reproduce this issue locally or on CI (https://github.com/ros2/ci/compare/env_lc_all).
As I don't remember any of the ROS 1 instructions requiring the locale to be set explicitly, so it must be an issue on the ROS 2 parsing side.
Looking at ament package it looks like it's already specifying `utf-8` when reading the files: https://github.com/ament/ament_package/pull/40.
Can you try that branch https://github.com/ament/ament_cmake/tree/locale_utf8_attempt and let me know if you see a change in behavior?

firesurfer, 2018-04-18:
Okey, now I can't reproduce this bug anymore even with the master branch. I guess it might be due to an update of my system. Apparently the python-lxml lib got updated (I'm not sure if that lib is used in the parser)

That might be the reason why you couldn't reproduce the bug.

Thanks for your help I'm closing this bug.

closed 2018-04-18 by firesurfer

richiware, 2018-04-19:
My Raspberry Pi has this problem. Using lastest branches in ROS2 I get this error:

```bash
+++ Building 'rmw_fastrtps_cpp'
==> '. /home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp'
/usr/bin/cmake -H/home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp -B/home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp --check-build-system CMakeFiles/Makefile.cmake 0
Re-run cmake file: Makefile older than: /home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml
-- Found ament_cmake_ros: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/ament_cmake_ros/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rcutils: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found rmw: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rmw/cmake)
-- Found rosidl_typesupport_introspection_c: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rosidl_typesupport_introspection_c/cmake)
-- Found rosidl_typesupport_introspection_cpp: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rosidl_typesupport_introspection_cpp/cmake)
-- Found ament_lint_auto: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/ament_lint_auto/cmake)
Error parsing '/home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml':
Traceback (most recent call last):
  File "/home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 145, in <module>
    main()
  File "/home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 56, in main
    raise e
  File "/home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 53, in main
    package = parse_package_string(args.package_xml.read())
  File "/usr/lib/python3.5/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 409: ordinal not in range(128)
CMake Error at /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/usr/bin/python3
  /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml
  /home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /home/pi/Workspace/ros2_ws/install/share/ament_lint_auto/cmake/ament_lint_auto_find_test_dependencies.cmake:30 (ament_package_xml)
  CMakeLists.txt:114 (ament_lint_auto_find_test_dependencies)
```

Using your change in ament_cmake I get a different error:

```bash
+++ Building 'rmw_fastrtps_cpp'
==> '. /home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp/cmake__build.sh && /usr/bin/make cmake_check_build_system' in '/home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp'
/usr/bin/cmake -H/home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp -B/home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp --check-build-system CMakeFiles/Makefile.cmake 0
Re-run cmake file: Makefile older than: /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package.cmake
-- Found ament_cmake_ros: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/ament_cmake_ros/cmake)
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found rcutils: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rcutils/cmake)
-- Found fastrtps_cmake_module: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found rmw: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rmw/cmake)
-- Found rosidl_typesupport_introspection_c: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rosidl_typesupport_introspection_c/cmake)
-- Found rosidl_typesupport_introspection_cpp: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/rosidl_typesupport_introspection_cpp/cmake)
-- Found ament_lint_auto: 0.4.0 (/home/pi/Workspace/ros2_ws/install/share/ament_lint_auto/cmake)
Traceback (most recent call last):
  File "/home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 145, in <module>
    main()
  File "/home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 62, in main
    f.write('%s\n' % line)
UnicodeEncodeError: 'ascii' codec can't encode character '\xe1' in position 45: ordinal not in range(128)
CMake Error at /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:94 (message):
  execute_process(/usr/bin/python3
  /home/pi/Workspace/ros2_ws/install/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/pi/Workspace/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/package.xml
  /home/pi/Workspace/ros2_ws/build/rmw_fastrtps_cpp/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
[... cut ...]
diff --git a/ament_cmake_core/cmake/core/package_xml_2_cmake.py b/ament_cmake_core/cmake/core/package_xml_2_cmake.py
index a910b00..b064489 100644
--- a/ament_cmake_core/cmake/core/package_xml_2_cmake.py
+++ b/ament_cmake_core/cmake/core/package_xml_2_cmake.py
@@ -57,7 +57,7 @@ def main(argv=sys.argv[1:]):

     lines = generate_cmake_code(package)
     if args.outfile:
-        with open(args.outfile, 'w') as f:
+       with open(args.outfile, 'w', encoding='UTF-8') as f:
             for line in lines:
                 f.write('%s\n' % line)
     else:
```

mikaelarguedas, 2018-05-28:
Thanks @richiware for trying it out, and suggesting an additional fix! Sorry I missed this comment for some reason.

My understanding is that both the patch [suggested here](https://github.com/ament/ament_cmake/commit/3d3c02b26948aa3708a3d2d0a924aa2c61a26cb5) and the one you posted here are necessary. Is that correct ?

mikaelarguedas, 2018-05-30:
Ok we will add my patch to https://github.com/ament/ament_cmake/pull/136 (that contains only your patch at the moment) then.

Thanks for testing and providing a patch!