ros2/rmw_fastrtps #219: Question about handling of array messages

anhosi, 2018-07-26:
I am not sure whether this is the correct place for this issue. So please direct me to a better location.

Together with @Karsten1987 we stumbled over
https://github.com/ros2/rmw_fastrtps/blob/8f6a13c5995b13a8bd9a0d147851ab5754bdf104/rmw_fastrtps_dynamic_cpp/include/rmw_fastrtps_dynamic_cpp/TypeSupport_impl.hpp#L279.

I have only limited understanding about the whole rmw. As far as I understand a array message member that is originally represented as a `std::vector<T> *` is `reinterpret_cast`ed to a `std::vector<unsigned char> *` with a `void *` as intermediate. IMHO this is not guaranteed to work by the C++ standard as implementations are allowed to use different template specializations of `std::vector` for different types. There are only guarantees for the memory layout of a vectors `data()`.
As far as I can tell the above code is only the symptom as the `void * field` here is part of what is passed into `rmw_serialize` as `ros_message`. Maybe it would be better to pass only the `data()` pointer of the vector at some point instead of the whole vector as `void *`?

deeplearningrobotics, 2018-07-26:
IMHO, if `field` was not originally  cast to `void *` from a `std::vector<unsigned char> *` then  this code yields undefined behavior according to the type aliasing rules.

https://en.cppreference.com/w/cpp/language/reinterpret_cast -> Type aliasing

anhosi, 2018-07-27:
@deeplearningrobotics exactly. From my understanding `field` might very well have been a `std::vector<float>` for a `float32[]` message member.

sloretz, 2018-08-09:
@richiprosima may I ask you to look at what it would take to fix this?

deeplearningrobotics, 2018-11-28:
@richiware: Could you take a look a this?

Or is this covered under the standard?

Chapter 3.10 (http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2013/n3690.pdf )

richiware, 2018-11-29:
I will try to make the changes next week.

deeplearningrobotics, 2018-11-29:
@richiware: There are a lot of `reinterpret_casts` and also casts to and from `void *` which are undocumented. So I think there is a generally bigger work packages to verify their correctness and somehow document that those do not exhibit UB.

ivanpauno, 2020-09-08:
Closing in favor of https://github.com/ros2/rmw_fastrtps/issues/428.
https://github.com/ros2/rmw_fastrtps/pull/429 is fixing this.

closed 2020-09-08 by ivanpauno