ros2/rmw_fastrtps #226: New test failure with latest Fast-RTPS

nuclearsandwich, 2018-09-05:
## Bug report

I'm not sure if this is a regression in Fast-RTPS or an update required in rmw_fastrtps_cpp. 

- Operating System:
  All Platforms

- Installation type:
  Source build

- Version or commit hash:
  First occurrence with https://github.com/eProsima/Fast-RTPS/commit/7b2c5b90ce626604b5f8a80b16c946b8d07552f8

- DDS implementation:
  rmw_fastrtps_cpp

#### Steps to reproduce issue

Reproducible on https://ci.ros2.org (See [this build](https://ci.ros2.org/job/ci_linux/5136/) for an example). To build locally you can use [this repos file](https://gist.github.com/nuclearsandwich/b9cb55e4809dd88fc80ac6e41984abdb).

#### Expected behavior

All rcl tests pass with rmw_fastrtps_cpp and Fast-RTPS on `master`

#### Actual behavior

The rcl test `rcl.TestClientFixture.test_client_nominal` fails with the following: <details><summary>output</summary>
<code><pre>
-- run_test.py: extra environment variables:
 - RMW_IMPLEMENTATION=rmw_fastrtps_cpp
-- run_test.py: extra environment variables to append:
 - LD_LIBRARY_PATH+=/home/rosbuild/ci_scripts/ws/build/rcl
-- run_test.py: invoking following command in '/home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl':
 - /home/rosbuild/ci_scripts/ws/build/rcl/test/test_client__rmw_fastrtps_cpp --gtest_output=xml:/home/rosbuild/ci_scripts/ws/build/rcl/test_results/rcl/test_client__rmw_fastrtps_cpp.gtest.xml
Running main() from gtest_main.cc
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from TestClientFixture
[ RUN      ] TestClientFixture.test_client_nominal
/home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl/test/rcl/test_client.cpp:88: Failure
Expected equality of these values:
  sequence_number
    Which is: 0
  1
[  FAILED  ] TestClientFixture.test_client_nominal (30 ms)
[ RUN      ] TestClientFixture.test_client_init_fini
[rcutils|error_handling.c:109] failed to allocate memory for the error state struct
[       OK ] TestClientFixture.test_client_init_fini (15 ms)
[----------] 2 tests from TestClientFixture (45 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (45 ms total)
[  PASSED  ] 1 test.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestClientFixture.test_client_nominal

 1 FAILED TEST
-- run_test.py: return code 1
</pre></code></details>

clalancette, 2018-09-07:
I did a bit of digging here, and it looks like https://github.com/eProsima/Fast-RTPS/commit/7b2c5b90ce626604b5f8a80b16c946b8d07552f8 is the culprit here.  What happens is that https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_client.cpp#L87 calls `rcl_send_request()` and expects the sequence number to be incremented by 1 at the end, but this is not happening.  The sequence goes:

https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_client.cpp#L87 ->
https://github.com/ros2/rcl/blob/master/rcl/src/rcl/client.c#L266 ->
https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_cpp/src/rmw_request.cpp#L25 ->
https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/rmw_request.cpp#L33 ->
https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/publisher/Publisher.cpp#L42 ->
https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/publisher/PublisherImpl.cpp#L83

(I gave up at this point and looked at the recent commits to Fast-RTPS).  @richiware , could you take a look?  Is this a problem with Fast-RTPS itself, or is the test expecting something incorrect?

mikaelarguedas, 2018-09-10:
@clalancette can you please provide permalinks instead of links pointing to the master branch ?

nuclearsandwich, 2018-09-11:
Here's a permalinked version of @clalancette's trace:

* https://github.com/ros2/rcl/blob/a0f83b071a67b1538f2210a1265fbb814aa0abd7/rcl/test*/rcl/test_client.cpp#L87
* https://github.com/ros2/rcl/blob/a0f83b071a67b1538f2210a1265fbb814aa0abd7/rcl/src/rcl/client.c#L266
* https://github.com/ros2/rmw_fastrtps/blob/e9c8b6915bc158848d16ecff66af91b70c02fd22/rmw_fastrtps_cpp/src/rmw_request.cpp#L25?noexpand
* https://github.com/ros2/rmw_fastrtps/blob/e9c8b6915bc158848d16ecff66af91b70c02fd22/rmw_fastrtps_shared_cpp/src/rmw_request.cpp#L33?noexpand
* https://github.com/eProsima/Fast-RTPS/blob/7b2c5b90ce626604b5f8a80b16c946b8d07552f8/src/cpp/publisher/Publisher.cpp#L42
* https://github.com/eProsima/Fast-RTPS/blob/7b2c5b90ce626604b5f8a80b16c946b8d07552f8/src/cpp/publisher/PublisherImpl.cpp#L83

richiware, 2018-09-11:
I've found the problem and created eProsima/Fast-RTPS#258 PR to solve it. I've checked against ros2 but if you can do it too, I will be more relaxed.

clalancette, 2018-09-11:
> I've found the problem and created eProsima/Fast-RTPS#258 PR to solve it. I've checked against ros2 but if you can do it too, I will be more relaxed.

@richiware, thanks, will take a look and report back.

nuclearsandwich, 2018-09-11:
CI of just the rcl tests with the hotfix branch

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

mikaelarguedas, 2018-09-11:
can you please also run CI with the system tests (test_communication, test_security and test_rclcpp) to make sure there is no side effect at the system level ?

mikaelarguedas, 2018-09-11:
CI including system tests:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

dirk-thomas, 2018-09-12:
With the upstream patch merged this can be closed.

closed 2018-09-12 by dirk-thomas