ros2/rmw_fastrtps #239: Fail to communicate service/client

DongheeYe, 2018-11-20:
## Bug report

**Required Info:**

- Operating System: Ubuntu 16.04
  - 
- Installation type: Source
  - 
- Version or commit hash: master
  - 
- DDS implementation: Fast-RTPS
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

1. Compile latest ros2 from master: colcon build --symlink-install --cmake-args -DSECURITY=ON -DCMAKE_BUILD_TYPE=Debug
2. Source the workspace: source install/local_setup.bash
3. Set envirionments
    ROS_SECURITY_ROOT_DIRECTORY=~/security/keys
    ROS_SECURITY_ENABLE=true
    ROS_SECURITY_STRATEGY=Permissive
4. Run service_main
```
    $ ./install/examples_rclcpp_minimal_service/lib/examples_rclcpp_minimal_service/service_main &
```
5. Run client_main
```
    $ ./install/examples_rclcpp_minimal_client/lib/examples_rclcpp_minimal_client/client_main
```

#### Expected behavior
The service and client can communicate properly.

#### Actual behavior
The client send request and waiting forever. (1 in 3 times)

#### Additional information
It is also occurred in -DSECURITY=OFF. But you can easily reproduce the issue if you build with -DSERCURITY=ON.

----
## Feature request

#### Feature description

#### Implementation considerations

DongheeYe, 2018-11-20:
I figure out the reason and fix it with https://github.com/ros2/rmw_fastrtps/pull/238.
Please check the patch.

dirk-thomas, 2019-01-11:
Closing since #238 is supposed to address this.

closed 2019-01-11 by dirk-thomas