ros2/rmw_fastrtps #246: Fast-RTPS and Connext to not communicate any longer

wjwwood, 2018-12-13:
## Bug report

**Required Info:**

- Operating System:
  - Windows (only)
- Installation type:
  - source
- Version or commit hash:
  - master (Fast-RTPS 1.7.0 and Connext 5.3.1)
- DDS implementation:
  - Fast-RTPS/Connext
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

In terminal A:

```
> call install\setup.bat
> set RMW_IMPLEMENTATION=rmw_connext_cpp
> install\demo_nodes_cpp\lib\demo_nodes_cpp\talker.exe
```

In terminal B:

```
> call install\setup.bat
> set RMW_IMPLEMENTATION=rmw_fastrtps_cpp
> install\demo_nodes_cpp\lib\demo_nodes_cpp\listener.exe
```

#### Expected behavior

They communicate and the listener receives data from the talker.

#### Actual behavior

Nothing is received by the listener.

#### Additional information

This occurs when swapping which is using Fast-RTPS/Connext (talker vs listener), and is resolved if you use either Fast-RTPS or Connext on both sides.

This is likely the root cause of new failures in our `test_communication` tests which do cross-vendor testing.

Screenshots:

[image]

[image]

richiware, 2018-12-13:
Before integrating FastRTPS v1.7.0 and the changes in rmw_fastrtps to support this version, they were tested against your CI ( [link](https://github.com/ros2/rmw_fastrtps/pull/233#issuecomment-444586237) ). As far as I know your CI checks communication between Fast RTPS and Connext. Do you know what changes are made in rmw_fastrtps after merging v1.7.0?

richiware, 2018-12-13:
We are trying to help investigating which could be the problem. We were analyzing your CI jobs. There is something we don't understand. We don't know how you CI jobs are internally and surely there is a reason. Why did job [1024](https://ci.ros2.org/view/nightly/job/nightly_win_rel/1024/) fail but next day job [1025](https://ci.ros2.org/view/nightly/job/nightly_win_rel/1025/) work successfully? It seems they used the same configuration and there are no significant changes in involved repositories. Is there something we don't get? some configuration that change between nightly jobs? Thanks.

nuclearsandwich, 2018-12-13:
While looking at last night's jobs I saw that [1042](https://ci.ros2.org/view/nightly/job/nightly_linux_debug/1042/testReport/) also has communication issues between connext and fastrtps. These failures are logged in https://github.com/ros2/build_cop/issues/153 and look to have first appeared in [1035](https://ci.ros2.org/view/nightly/job/nightly_linux_debug/1035/testReport).

> Why did job [1024](https://ci.ros2.org/view/nightly/job/nightly_win_rel/1024/) fail but next day job [1025](https://ci.ros2.org/view/nightly/job/nightly_win_rel/1025/) work successfully? It seems [...]

It's not an explanatory reason, but our CI defaults to re-running failed tests up to 10 times to see if they'll pass. This is a mitigation against tests that may flake due to network or other variable conditions but it also muddies the waters when trying to pinpoint the exact start of issues that don't occur every time. That the tests sometimes fail and sometimes don't suggest the issue is not reliably reproduced. Although it has certainly become more reliable to reproduce on Windows and possibly in debug configurations on Linux now as well.

MiguelCompany, 2018-12-13:
As I have stated [here](https://github.com/ros2/ros2/issues/621#issuecomment-446929472) it seems connext is now too restrictive on guidPrefix values. Could you check with eProsima/Fast-RTPS#353 ?

nuclearsandwich, 2018-12-13:
Thanks for linking that @MiguelCompany. I've triggered a build of our communication tests with the retest-until-pass setting reduced from 10 to 3. 

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

nuclearsandwich, 2018-12-13:
There is warning output during cross-communication reported in https://github.com/ros2/demos/issues/293 but (Fast-RTPS <-> Connext is toward the end of the description). It doesn't appear that communication was inhibited so those warnings may or may not be related.

MiguelCompany, 2018-12-14:
@nuclearsandwich @wjwwood On the Linux debug build, I see that Connext is failing to initialize the rcl node on some tests where Fast-RTPS is not involved.

For instance, [this test](https://ci.ros2.org/job/ci_linux/5893/testReport/junit/(root)/projectroot/test_messages_c__rmw_connext_cpp/) says
>[D0108|ENABLE]DDS_DomainParticipant_enableI:Automatic participant index failed to initialize. PLEASE VERIFY CONSISTENT TRANSPORT / DISCOVERY CONFIGURATION. 
DDSDomainParticipant_impl::createI:ERROR: Failed to auto-enable entity
DomainParticipantFactory_impl::create_participant():!create failure creating participant
/home/rosbuild/ci_scripts/ws/src/ros2/system_tests/test_communication/test/test_messages_c.cpp:110: Failure
Expected equality of these values:
  0
  ret
    Which is: 1
failed to create participant, at /home/rosbuild/ci_scripts/ws/src/ros2/rmw_connext/rmw_connext_shared_cpp/src/node.cpp:261, at /home/rosbuild/ci_scripts/ws/src/ros2/rcl/rcl/src/rcl/node.c:401

MiguelCompany, 2018-12-14:
@nuclearsandwich @wjwwood FYI, eProsima/Fast-RTPS#353 has been merged on master.

mjcarroll, 2018-12-18:
During the lead-up to Crystal, I tested cross-vendor support on my Windows 10 Virtual Machine, and did not find any issues.

In one of the hangouts, @wjwwood put forward the hypothesis that FastRTPS, OpenSplice, and Connext may be choosing different network interfaces in order to do discovery or connectivity, leading the nodes to fail to discover each other.  This is only a hypothesis and has not been validated or researched in any way.

richiware, 2019-02-13:
We fixed an issue sending multicast in a Windows machine with several interfaces and some of them disconnected. We think that issue should fix this one. Can you confirm?

richiware, 2019-02-20:
[v1.7.1](https://github.com/eProsima/Fast-RTPS/releases/tag/v1.7.1) incorporates eProsima/Fast-RTPS#394. Can you verify your nightly job works as expected? Thanks

MiguelCompany, 2019-10-24:
I think this can be closed now that ros2/ros2#814 has been merged

closed 2019-11-15 by dirk-thomas