ros2/rmw_fastrtps #251: Bind a Fast RTPS publisher with a ROS 2 subscriber fails on ROS2 Bouncy

TSC21, 2018-12-31:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - latest release for `bouncy`
- DDS implementation:
  - Fast RTPS

#### Steps to reproduce issue
Follow the steps set in:
1. https://dev.px4.io/en/middleware/micrortps.html#setting-up-the-workspaces
2. https://dev.px4.io/en/middleware/micrortps.html#building-the-workspaces
3. https://dev.px4.io/en/middleware/micrortps.html#testing-the-px4-fastrpts-bridge-with-ros2-and-ros

The above will generate sources and headers for a micro RTPS agent which will publish and subscribe to/from data coming from the ROS2 Fast RTPS layer.

#### Expected behavior
With Ardent, the agent is able to bind with ROS2 publishers/subscribers - a `Publisher matched` or a `Subscriber matched` message appears on the console output of the agent.

#### Actual behavior
Currently it only appears to work using Ardent.

#### Additional information

It's not obvious to me if this is related to the current implementation of the `rmw_fastrtps` to Bouncy, but it was proven that this problem only occurs while using Bouncy. The bind happens when using Ardent.

The current template used for the micro-RTPS agent Publisher is https://github.com/PX4/px4_ros_com/blob/master/msg/templates/Publisher.cpp.template#L76-L104. Again, this template works well with Ardent and allows binding the agent and the ROS2 nodes.
I tried to research the current implementation of Fast RTPS publishers and subscribers inside `rmw_fastrtps` but could not find any relevant differences that would limit the functionality.

Any tips to solve this are welcomed!

Thanks in advance!

TSC21, 2019-01-05:
@nuclearsandwich @dirk-thomas any input you can give here? Thanks!

dirk-thomas, 2019-01-05:
In the code snippet creating the publisher you are setting a partition. In Ardent the namespace part of a topic was mapped to DDS partitions. In Bouncy as well as Crystal that is not the case anymore. Partitions are not used anything - instead the namespace stays part of the topic name.

Not sure if the following is the best reference but it should give you a starting point: ros2/ros2#476

TSC21, 2019-01-07:
@dirk-thomas I just tried to remove the partitions and the result is the same. Here's an example of a generated message Publisher:
```c++
#include <fastrtps/participant/Participant.h>
#include <fastrtps/attributes/ParticipantAttributes.h>
#include <fastrtps/publisher/Publisher.h>
#include <fastrtps/attributes/PublisherAttributes.h>

#include <fastrtps/Domain.h>

#include <fastrtps/utils/eClock.h>

#include "SensorCombined_Publisher.h"

using namespace eprosima::fastrtps;
using namespace eprosima::fastrtps::rtps;

SensorCombined_Publisher::SensorCombined_Publisher() : mp_participant(nullptr), mp_publisher(nullptr) {}

SensorCombined_Publisher::~SensorCombined_Publisher() { Domain::removeParticipant(mp_participant);}

bool SensorCombined_Publisher::init()
{
    // Create RTPSParticipant

    ParticipantAttributes PParam;
    PParam.rtps.builtin.domainId = 0;
    PParam.rtps.builtin.leaseDuration = c_TimeInfinite;
    PParam.rtps.setName("SensorCombined_publisher");  //You can put here the name you want
    mp_participant = Domain::createParticipant(PParam);
    if(mp_participant == nullptr)
        return false;

    //Register the type

    Domain::registerType(mp_participant, (TopicDataType*) &myType);

    // Create Publisher

    PublisherAttributes Wparam;
    Wparam.topic.topicKind = NO_KEY;
    Wparam.topic.topicDataType = myType.getName();  //This type MUST be registered
    Wparam.topic.topicName = "SensorCombined_topic";
    mp_publisher = Domain::createPublisher(mp_participant, Wparam, (PublisherListener*) &m_listener);
    if(mp_publisher == nullptr)
        return false;
    //std::cout << "Publisher created, waiting for Subscribers." << std::endl;
    return true;
}

void SensorCombined_Publisher::PubListener::onPublicationMatched(Publisher* pub, MatchingInfo& info)
{
    if (info.status == MATCHED_MATCHING)
    {
        n_matched++;
        std::cout << "Publisher matched" << std::endl;
    }
    else
    {
        n_matched--;
        std::cout << "Publisher unmatched" << std::endl;
    }
}

void SensorCombined_Publisher::run()
{
    while(m_listener.n_matched == 0)
    {
        eClock::my_sleep(250); // Sleep 250 ms
    }

[... 125 lines cut ...]
                p_type->serializeKey(ser);
                if(SensorCombined_::getKeyMaxCdrSerializedSize()>16)	{
                    m_md5.init();
                    m_md5.update(m_keyBuffer,(unsigned int)ser.getSerializedDataLength());
                    m_md5.finalize();
                    for(uint8_t i = 0;i<16;++i)    	{
                        handle->value[i] = m_md5.digest[i];
                    }
                }
                else    {
                    for(uint8_t i = 0;i<16;++i)    	{
                        handle->value[i] = m_keyBuffer[i];
                    }
                }
                return true;
            }

        } //End of namespace dds_

    } //End of namespace msg

} //End of namespace px4_ros_com
```

The reference you gave me seems to address the removal of the partition, but I don't know what else should I aim to here. Any other tip? Thanks!

dirk-thomas, 2019-01-07:
Your snippet uses the topic name `SensorCombined_topic`. I don't know what you are using on the ROS 2 side as the topic name but after name mangling I would expect it to not be equal to `SensorCombined_topic`.

I would suggest to either debug the subscriber on the ROS 2 side to see what actual values are being used (so that your code can match all names and QoS settings) or to use a tool to introspect the RTPS packets which will show how each side tried to discover the publisher / subscriber and find the difference between them that way.

TSC21, 2019-01-07:
The listener code is exactly the same for Ardent and for Bouncy, though just works on Ardent: https://github.com/PX4/px4_ros_com/blob/master/src/listeners/sensor_combined_listener.cpp. The topic naming matches. Anything abnormal you can see on it?
Can you suggest a tool for do that introspection?

dirk-thomas, 2019-01-07:
If you have access to the RTI Admin Console I would recommend that. Otherwise Wireshark is probably your best bet.

TSC21, 2019-01-07:
Thanks I will try that. Though, can you specify what were the more relevant changes on the rmw_fastrtps implementation from Ardent to Bouncy? I mean, what differences should I expect to find besides the partition? Anything on the QoS as well?

dirk-thomas, 2019-01-07:
You can find all changes between Bouncy and Ardent here: https://github.com/ros2/rmw_fastrtps/compare/ardent...bouncy

Beside the already mentioned ros2/rmw_fastrtps#192 nothing sticks out to me.

As mentioned before I would suggest to double check your code. Before you used `rt` as the partition but there is no reference of `rt` in the new snippet. While the partitions have been dropped their content still needs to be part of the actual topic name.

TSC21, 2019-01-07:
> While the partitions have been dropped their content still needs to be part of the actual topic name.

I am not sure of the actual meaning of this. Does that mean I need to include the partition but pass it with an empty string?

dirk-thomas, 2019-01-07:
Your topic name needs to be something like this `/rt/SensorCombined_topic` to match the mangling happening in ROS 2. Since I am not sure about the exact value out of my head I suggest to print the value used in ROS 2 FastRTPS.

wjwwood, 2019-01-07:
You should be able to see the exact names with this script:

```python
#!/usr/bin/env python3

import pprint

import rclpy

rclpy.init()
n = rclpy.create_node('list_no_demangle')
pprint.pprint(n.get_topic_names_and_types(True))

```

For example with talker I get this:

```
[('rq/_ros2cli_node_daemon_87/describe_parametersRequest',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Request_']),
 ('rq/_ros2cli_node_daemon_87/get_parameter_typesRequest',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Request_']),
 ('rq/_ros2cli_node_daemon_87/get_parametersRequest',
  ['rcl_interfaces::srv::dds_::GetParameters_Request_']),
 ('rq/_ros2cli_node_daemon_87/list_parametersRequest',
  ['rcl_interfaces::srv::dds_::ListParameters_Request_']),
 ('rq/_ros2cli_node_daemon_87/set_parametersRequest',
  ['rcl_interfaces::srv::dds_::SetParameters_Request_']),
 ('rq/_ros2cli_node_daemon_87/set_parameters_atomicallyRequest',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Request_']),
 ('rq/list_no_demangle/describe_parametersRequest',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Request_']),
 ('rq/list_no_demangle/get_parameter_typesRequest',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Request_']),
 ('rq/list_no_demangle/get_parametersRequest',
  ['rcl_interfaces::srv::dds_::GetParameters_Request_']),
 ('rq/list_no_demangle/list_parametersRequest',
  ['rcl_interfaces::srv::dds_::ListParameters_Request_']),
 ('rq/list_no_demangle/set_parametersRequest',
  ['rcl_interfaces::srv::dds_::SetParameters_Request_']),
 ('rq/list_no_demangle/set_parameters_atomicallyRequest',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Request_']),
 ('rq/talker/describe_parametersRequest',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Request_']),
 ('rq/talker/get_parameter_typesRequest',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Request_']),
 ('rq/talker/get_parametersRequest',
  ['rcl_interfaces::srv::dds_::GetParameters_Request_']),
 ('rq/talker/list_parametersRequest',
  ['rcl_interfaces::srv::dds_::ListParameters_Request_']),
 ('rq/talker/set_parametersRequest',
  ['rcl_interfaces::srv::dds_::SetParameters_Request_']),
 ('rq/talker/set_parameters_atomicallyRequest',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Request_']),
 ('rr/_ros2cli_node_daemon_87/describe_parametersReply',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Response_']),
 ('rr/_ros2cli_node_daemon_87/get_parameter_typesReply',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Response_']),
 ('rr/_ros2cli_node_daemon_87/get_parametersReply',
  ['rcl_interfaces::srv::dds_::GetParameters_Response_']),
 ('rr/_ros2cli_node_daemon_87/list_parametersReply',
  ['rcl_interfaces::srv::dds_::ListParameters_Response_']),
 ('rr/_ros2cli_node_daemon_87/set_parametersReply',
  ['rcl_interfaces::srv::dds_::SetParameters_Response_']),
 ('rr/_ros2cli_node_daemon_87/set_parameters_atomicallyReply',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Response_']),
 ('rr/list_no_demangle/describe_parametersReply',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Response_']),
 ('rr/list_no_demangle/get_parameter_typesReply',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Response_']),
 ('rr/list_no_demangle/get_parametersReply',
  ['rcl_interfaces::srv::dds_::GetParameters_Response_']),
 ('rr/list_no_demangle/list_parametersReply',
  ['rcl_interfaces::srv::dds_::ListParameters_Response_']),
 ('rr/list_no_demangle/set_parametersReply',
  ['rcl_interfaces::srv::dds_::SetParameters_Response_']),
 ('rr/list_no_demangle/set_parameters_atomicallyReply',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Response_']),
 ('rr/talker/describe_parametersReply',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Response_']),
 ('rr/talker/get_parameter_typesReply',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Response_']),
 ('rr/talker/get_parametersReply',
  ['rcl_interfaces::srv::dds_::GetParameters_Response_']),
 ('rr/talker/list_parametersReply',
  ['rcl_interfaces::srv::dds_::ListParameters_Response_']),
 ('rr/talker/set_parametersReply',
  ['rcl_interfaces::srv::dds_::SetParameters_Response_']),
 ('rr/talker/set_parameters_atomicallyReply',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Response_']),
 ('rt/chatter', ['std_msgs::msg::dds_::String_']),
 ('rt/parameter_events', ['rcl_interfaces::msg::dds_::ParameterEvent_'])]
```

Note this:

```
 ('rt/chatter', ['std_msgs::msg::dds_::String_']),
```

TSC21, 2019-01-07:
This is fantastic! Thank you very much @dirk-thomas (and @wjwwood for the script)! I just added the `rt/` suffix to the topic name and it worked as it should. This can be closed now!

TSC21, 2019-03-04:
@dirk-thomas, @wjwwood I am facing again the same issue regarding the binding between a Fast-RTPS publisher and a ROS subscriber, but this time with ROS2 Crystal.
The publisher now has the required suffix, as you can see on the bellow code:
```c++
bool SensorCombined_Publisher::init()
{
    // Create RTPSParticipant

    ParticipantAttributes PParam;
    PParam.rtps.builtin.domainId = 0;
    PParam.rtps.builtin.leaseDuration = c_TimeInfinite;
    PParam.rtps.setName("SensorCombined_publisher");  //You can put here the name you want
    mp_participant = Domain::createParticipant(PParam);
    if(mp_participant == nullptr)
        return false;

    //Register the type

    Domain::registerType(mp_participant, (TopicDataType*) &myType);

    // Create Publisher

    PublisherAttributes Wparam;
    Wparam.topic.topicKind = NO_KEY;
    Wparam.topic.topicDataType = myType.getName();  //This type MUST be registered
    Wparam.topic.topicName = "rt/SensorCombined_topic";
    mp_publisher = Domain::createPublisher(mp_participant, Wparam, (PublisherListener*) &m_listener);
    if(mp_publisher == nullptr)
        return false;
    //std::cout << "Publisher created, waiting for Subscribers." << std::endl;
    return true;
}
```
And the respective ROS2 subscriber:
```c++
#include <rclcpp/rclcpp.hpp>
#include <px4_msgs/msg/sensor_combined.hpp>

/**
 * @brief Sensor Combined uORB topic data callback
 */
class SensorCombinedListener : public rclcpp::Node
{
public:
	explicit SensorCombinedListener() : Node("sensor_combined_listener") {
		subscription_ = this->create_subscription<px4_msgs::msg::SensorCombined>(
		"SensorCombined_topic",
		[this](const px4_msgs::msg::SensorCombined::UniquePtr msg) {
			std::cout << "\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n\n";
			std::cout << "RECEIVED DATA ON SENSOR COMBINED"   << std::endl;
			std::cout << "================================"   << std::endl;
			std::cout << "ts: "	     << msg->timestamp    << std::endl;
			std::cout << "gyro_rad[0]: " << msg->gyro_rad[0]  << std::endl;
			std::cout << "gyro_rad[1]: " << msg->gyro_rad[1]  << std::endl;
			std::cout << "gyro_rad[2]: " << msg->gyro_rad[2]  << std::endl;
			std::cout << "gyro_integral_dt: " << msg->gyro_integral_dt << std::endl;
			std::cout << "accelerometer_timestamp_relative: " << msg->accelerometer_timestamp_relative << std::endl;
			std::cout << "accelerometer_m_s2[0]: " << msg->accelerometer_m_s2[0] << std::endl;
			std::cout << "accelerometer_m_s2[1]: " << msg->accelerometer_m_s2[1] << std::endl;
			std::cout << "accelerometer_m_s2[2]: " << msg->accelerometer_m_s2[2] << std::endl;
			std::cout << "accelerometer_integral_dt: " << msg->accelerometer_integral_dt << std::endl;
		});
	}

private:
	rclcpp::Subscription<px4_msgs::msg::SensorCombined>::SharedPtr subscription_;
};

int main(int argc, char *argv[])
{
	std::cout << "Starting sensor_combined listener node..." << std::endl;
	setvbuf(stdout, NULL, _IONBF, BUFSIZ);
	rclcpp::init(argc, argv);
	rclcpp::spin(std::make_shared<SensorCombinedListener>());

	rclcpp::shutdown();
	return 0;
}
```
Is there any QoS configuration, or something else, that I am missing in comparison with the working setup on Bouncy? Thanks in advance!

TSC21, 2019-03-04:
@wjwwood btw, using your script I get the following output:
```sh
[('rq/list_no_demangle/describe_parametersRequest',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Request_']),
 ('rq/list_no_demangle/get_parameter_typesRequest',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Request_']),
 ('rq/list_no_demangle/get_parametersRequest',
  ['rcl_interfaces::srv::dds_::GetParameters_Request_']),
 ('rq/list_no_demangle/list_parametersRequest',
  ['rcl_interfaces::srv::dds_::ListParameters_Request_']),
 ('rq/list_no_demangle/set_parametersRequest',
  ['rcl_interfaces::srv::dds_::SetParameters_Request_']),
 ('rq/list_no_demangle/set_parameters_atomicallyRequest',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Request_']),
 ('rq/sensor_combined_listener/describe_parametersRequest',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Request_']),
 ('rq/sensor_combined_listener/get_parameter_typesRequest',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Request_']),
 ('rq/sensor_combined_listener/get_parametersRequest',
  ['rcl_interfaces::srv::dds_::GetParameters_Request_']),
 ('rq/sensor_combined_listener/list_parametersRequest',
  ['rcl_interfaces::srv::dds_::ListParameters_Request_']),
 ('rq/sensor_combined_listener/set_parametersRequest',
  ['rcl_interfaces::srv::dds_::SetParameters_Request_']),
 ('rq/sensor_combined_listener/set_parameters_atomicallyRequest',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Request_']),
 ('rr/list_no_demangle/describe_parametersReply',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Response_']),
 ('rr/list_no_demangle/get_parameter_typesReply',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Response_']),
 ('rr/list_no_demangle/get_parametersReply',
  ['rcl_interfaces::srv::dds_::GetParameters_Response_']),
 ('rr/list_no_demangle/list_parametersReply',
  ['rcl_interfaces::srv::dds_::ListParameters_Response_']),
 ('rr/list_no_demangle/set_parametersReply',
  ['rcl_interfaces::srv::dds_::SetParameters_Response_']),
 ('rr/list_no_demangle/set_parameters_atomicallyReply',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Response_']),
 ('rr/sensor_combined_listener/describe_parametersReply',
  ['rcl_interfaces::srv::dds_::DescribeParameters_Response_']),
 ('rr/sensor_combined_listener/get_parameter_typesReply',
  ['rcl_interfaces::srv::dds_::GetParameterTypes_Response_']),
 ('rr/sensor_combined_listener/get_parametersReply',
  ['rcl_interfaces::srv::dds_::GetParameters_Response_']),
 ('rr/sensor_combined_listener/list_parametersReply',
  ['rcl_interfaces::srv::dds_::ListParameters_Response_']),
 ('rr/sensor_combined_listener/set_parametersReply',
  ['rcl_interfaces::srv::dds_::SetParameters_Response_']),
 ('rr/sensor_combined_listener/set_parameters_atomicallyReply',
  ['rcl_interfaces::srv::dds_::SetParametersAtomically_Response_']),
 ('rt/AdcReport_topic', ['px4_ros_com::msg::dds_::AdcReport_']),
 ('rt/Airspeed_topic', ['px4_ros_com::msg::dds_::Airspeed_']),
 ('rt/BatteryStatus_topic', ['px4_ros_com::msg::dds_::BatteryStatus_']),
 ('rt/CameraCapture_topic', ['px4_ros_com::msg::dds_::CameraCapture_']),
 ('rt/CameraTrigger_topic', ['px4_ros_com::msg::dds_::CameraTrigger_']),
 ('rt/CollisionReport_topic', ['px4_ros_com::msg::dds_::CollisionReport_']),
 ('rt/Cpuload_topic', ['px4_ros_com::msg::dds_::Cpuload_']),
 ('rt/DebugArray_topic', ['px4_ros_com::msg::dds_::DebugArray_']),
 ('rt/DebugKeyValue_topic', ['px4_ros_com::msg::dds_::DebugKeyValue_']),
 ('rt/DebugValue_topic', ['px4_ros_com::msg::dds_::DebugValue_']),
 ('rt/DebugVect_topic', ['px4_ros_com::msg::dds_::DebugVect_']),
 ('rt/EstimatorStatus_topic', ['px4_ros_com::msg::dds_::EstimatorStatus_']),
 ('rt/HomePosition_topic', ['px4_ros_com::msg::dds_::HomePosition_']),
 ('rt/IridiumsbdStatus_topic', ['px4_ros_com::msg::dds_::IridiumsbdStatus_']),
 ('rt/ObstacleDistance_topic', ['px4_ros_com::msg::dds_::ObstacleDistance_']),
 ('rt/PositionSetpointTriplet_topic',
  ['px4_ros_com::msg::dds_::PositionSetpointTriplet_']),
 ('rt/SensorCombined_topic', ['px4_msgs::msg::dds_::SensorCombined_']),
 ('rt/SensorSelection_topic', ['px4_ros_com::msg::dds_::SensorSelection_']),
 ('rt/TrajectoryWaypoint_topic',
  ['px4_ros_com::msg::dds_::TrajectoryWaypoint_']),
 ('rt/VehicleAttitude_topic', ['px4_ros_com::msg::dds_::VehicleAttitude_']),
 ('rt/VehicleOdometry_topic', ['px4_ros_com::msg::dds_::VehicleOdometry_']),
 ('rt/VehicleTrajectoryWaypoint_topic',
  ['px4_ros_com::msg::dds_::VehicleTrajectoryWaypoint_']),
 ('rt/WindEstimate_topic', ['px4_ros_com::msg::dds_::WindEstimate_']),
 ('rt/parameter_events', ['rcl_interfaces::msg::dds_::ParameterEvent_']),
 ('rt/rosout', ['rcl_interfaces::msg::dds_::Log_'])]
```

wjwwood, 2019-03-05:
I see only `rt/SensorCombined_topic` in the output from your script, so it looks like it's wired correctly. You might want to check the other settings and ensure communication is working otherwise.

TSC21, 2019-03-08:
> I see only `rt/SensorCombined_topic` in the output from your script, so it looks like it's wired correctly. You might want to check the other settings and ensure communication is working otherwise.

@wjwwood thanks for the feedback. What other settings are these that made this incompatible from ROS Bouncy to ROS Crystal?

wjwwood, 2019-03-08:
There were none of which I am aware, that's why I encouraged you to double check that your network is working properly. Something may have changed, but not that I can remember.

TSC21, 2019-03-09:
> that's why I encouraged you to double check that your network is working properly.

How can that be verified? There are no changes on the agent code or the ROS2 subscriber that would justify this.

TSC21, 2019-03-09:
@richiware maybe you can give a hint here? Thanks!

richiware, 2019-03-11:
Can you provide me a Wireshark capture where FastRTPS publisher and ROS2 subscriber are involved? It will be helpful for me.

TSC21, 2019-03-17:
@richiware @richiprosima here it goes:https://drive.google.com/open?id=1bE9j-zO-2_4BcieeDeuxlY2Q9qOSZhPS.

On this example, you have the microRTPS agent publishing multiple types of messages, but the ones in specific that matter are the SensorCombined ones, for which the `sensor_combined_listener` (the ROS2 listener) is looking for. Any tips on how to solve this would help. Thanks!

TSC21, 2019-03-21:
@richiware can you help me with this? Thank you

richiware, 2019-03-21:
I miss some submessages. Can you capture also the localhost interface?
Also I see the ROS2 participant uses in its participant info the parameter UserData with the content `name=sensor_combined_listener;namespace=/;`. I don't know if this is important for ROS2 communications.

richiware, 2019-03-21:
I don't know, maybe @wjwwood could answer that.

wjwwood, 2019-03-21:
No, it will not prevent communication, but it is used for things like `ros2 node list`.

TSC21, 2019-03-27:
> No, it will not prevent communication, but it is used for things like `ros2 node list`.

Well, then I am kinda stuck on what to actually investigate.

TSC21, 2019-03-27:
@wjwwood @richiware can you please suggest on other things to verify so to understand why there's no bind? I really appreciate your inputs. Thanks in advance

wjwwood, 2019-03-27:
@TSC21 did you give @richiprosima the "capture data with localhost"? I don't have any more ideas, I haven't worked on either Fast-RTPS or rmw_fastrtps_cpp recently, but eProsima did refactor the rmw implementation last release (e.g. https://github.com/ros2/rmw_fastrtps/pull/203), so presumably they know if anything significant changed.

After that, communication works for us, and all the code that configures Fast-RTPS for us is here:

https://github.com/ros2/rmw_fastrtps/

So you can look at that yourself to see if we're doing anything strange, but I don't think we are.

TSC21, 2019-03-27:
> @TSC21 did you give @richiprosima the "capture data with localhost"?

I was wondering what in specific are we looking for on `localhost` besides RTPS messages, which was what I sent to @richiprosima. But yes I guess I can do a full capture.

> So you can look at that yourself to see if we're doing anything strange, but I don't think we are.

I already tried that myself but I nothing comes obvious to my eyes. The PR you are showing me has the code of the release I was already using in Bouncy, and I suppose now in Crystal. What I wanted to understand if it there was some obvious change on the participant or the publisher/subscribers configurations that need to be taken into account so there's a bind between a pure Fast-RTPS publisher and a ROS2 subscriber.

richiware, 2019-03-27:
@TSC21, as you are executing both applications in the same machine, several RTPS messages are sent through localhost, and I couldn't see them in your previous Wireshark capture.

TSC21, 2019-03-28:
> @TSC21, as you are executing both applications in the same machine, several RTPS messages are sent through localhost, and I couldn't see them in your previous Wireshark capture.

@richiware understood. Here it goes: https://drive.google.com/open?id=1WctMEIdQrokgFZrfBVTnYidszuZwCMNb. Thanks in advance!

TSC21, 2019-03-31:
@richiprosima @richiware any input? Thanks!

richiware, 2019-04-01:
Your Publisher and ROS2 Subscriber are using the same topic name: "rt/SensorCombined_topic". But the type name differs. Your Publisher is using "px4_ros_com::msg::dds_::SensorCombined_" and ROS2 Subscriber is using "px4_msgs::msg::dds_::SensorCombined_". This is the reason they are not matching.

TSC21, 2019-04-01:
Oh right!! That makes much more sense now! Can you please tell me how did yoi verify that on the Wireshark? After some inspection I didn't find that info. Thanks!!

richiware, 2019-04-01:
Publisher info (`DATA(w)`) and Subscriber info (`DATA(r)`)  have this information inside.

TSC21, 2019-04-08:
@richiware thank you very much for your input! I was able to solve this by setting the same userspace type for the publisher and the subscriber. It was an issue introduced after a big package restructure. Again, thanks!

TSC21, 2019-10-24:
@richiprosima looks like for each release I always stump into the same issue. This time with ROS 2 Dashing. Using Fast-RTPS 1.8.2. But now, trying to analyse the Wireshark capture, I didn't find any clues: https://drive.google.com/open?id=1QTx6xjRQZ1Rh1F-sibek40AvvSsF3hw6. Topic names look the same, types as well. I don't find `DATA(r)` or `DATA(w)` though. Only `DATA(p)`.

Can you help here? Thank you!

closed 2020-03-24 by TSC21