ros2/rmw_fastrtps #255: Ethernet connection crashing when running multiple nodes

alsora, 2019-02-05:
## Bug report

**Required Info:**

- Operating System:
 Ubuntu 16.04/18.04
- Installation type:
binaries
- Version or commit hash:
crystal
- DDS implementation:
Fast-RTPS
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

Issue related to this question https://answers.ros.org/question/313628/ros2-and-network-usage/

Whenever I'm connected using an ethernet cable and I start multiple nodes at the same time (launch file or from a C++ executable) the laptop's network gets disconnected.

It's something related to the nodes creation, as it happens even if nodes are not spinning/publishing.

It's something related to the type of connection as it doesn't happen if I'm connected to the wifi and it is not happening if I try the system connected to a different ethernet network.

It doesn't happen with OpenSplice or Connext DDS.

Here, the usual piece of code to reproduce the issue

```

#include <memory>
#include <iostream>
#include <thread>
#include <sstream>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/header.hpp"

using namespace std::chrono_literals;

class SimpleSubscriberNode : public rclcpp::Node {

public:

    SimpleSubscriberNode(std::string name) : Node(name)
    {
        _subscriber = this->create_subscription<std_msgs::msg::Header>("my_topic",
            std::bind(&SimpleSubscriberNode::simple_callback, this, std::placeholders::_1));

        msg_count = 0;
    }

    int get_count() {return msg_count;}

private:

    void simple_callback(const std_msgs::msg::Header::SharedPtr msg)
    {
        (void)msg;
        msg_count ++;
    }

    rclcpp::Subscription<std_msgs::msg::Header>::SharedPtr _subscriber;
    int msg_count;
};

int main(int argc, char ** argv)
{
    rclcpp::init(argc, argv);

    int n_subscribers = 20;
    int experiment_duration = 20;

    rclcpp::executors::SingleThreadedExecutor::SharedPtr executor =
        std::make_shared<rclcpp::executors::SingleThreadedExecutor>();

    std::vector<std::shared_ptr<SimpleSubscriberNode>> subscriber_nodes;

    for (int i = 0; i < n_subscribers; i++){
        std::stringstream ss;
        ss << "node_";
        ss << i;
        std::shared_ptr<SimpleSubscriberNode> sub_node = std::make_shared<SimpleSubscriberNode>(ss.str());

        subscriber_nodes.push_back(sub_node);
        executor->add_node(sub_node);
    }

    std::this_thread::sleep_for(std::chrono::seconds(experiment_duration));

    rclcpp::shutdown();

    return 0;

}

```

cottsay, 2019-02-14:
Have you tried reproducing this with other hardware? Possibly from another machine, or using a USB Ethernet adapter?

alsora, 2019-02-14:
Yes, I have 3 laptops and it happens on all of them.

The same laptops do not present the problem when I was connected to a different network.

Do you have any idea what should I look for, in order to debug this issue?

richiware, 2019-02-14:
I've executed your example in my linux laptop and couldn't reproduce your problem. Does it also occur with the HelloWorld example that comes with Fast RTPS?

alsora, 2019-02-14:
Do you mean this ?
https://github.com/eProsima/Fast-RTPS/tree/master/examples/C%2B%2B/HelloWorldExample

This is something different from my problem, as it's just creating one participant each time you call the executable. 

I have the problem if I start some nodes almost at the same time, as when using a launch file for example.

I noticed that I don't have this problem if I run a single node, even if it has a huge amount of subscribers/publishers.
May this be related to the discovery process?

richiware, 2019-02-20:
I will continue to reproduce your problem. Meantime can you test against new v1.7.1 released today? It will be very useful because we made improvements in the discovery phase.

alsora, 2019-02-20:
It's still present.
Unfortunately I really don't understand what can be the cause... If you have any idea or something that I can test, don't hesitate to ask.

alsora, 2019-03-13:
Hi, unfortunately I do not have new updates on this issue.

However I noticed that when I run a single ROS2 process with many participants, FastRTPS opens a different port for each of them.
Could this be related?

richiware, 2019-03-18:
I continue being not able to reproduce your issue.

How it is opening ports is normal. Several participants will use the same port for multicast metatraffic. But each one will open another two unicast ports, for unicast metatraffic and user data. This is the common behavior in DDS and I don't think this is related.

alsora, 2019-03-18:
Ok thank you a lot for your eplainations.
I think that we can close this issue since it is not reproducible.
In case I will re-open in once I get some relevant updates

closed 2019-03-18 by alsora

Leeez, 2023-03-27:
Hello, I'm sorry to disturb you after so many years. Thank you for your question. We have encountered the same problem. Have you solved this problem? @alsora