ros2/rmw_fastrtps #261: Max value for ROS_DOMAIN_ID

alsora, 2019-03-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - crystal
- DDS implementation:
  - FastRTPS 1.7.0

#### Steps to reproduce issue

This is linked to [this post on ROS Answers](https://answers.ros.org/question/318386/ros2-max-domain-id/).

```
export ROS_DOMAIN_ID=9999

ros2 run examples_rclcpp_minimal_publisher publisher_lambda
```

#### Expected behavior

The ROS_DOMAIN_ID should be used to decide which UDP port has to be opened for a participant.
Due to the limit of UDP ports, admissible values are in the range 0-232.

An error should be raised.

With FastRTPS, the only error related to this occur if the ROS_DOMAIN_ID exceeds the limit of `unsigned long` which is used in RCL for storing it.

#### Actual behavior

No error occurs and I'm able to communicate with another machine with the same ROS_DOMAIN_ID .

Trying the same thing with OpenSplice or Connext, a runtime error is raised.

dirk-thomas, 2019-03-13:
The FastRTPS API `eprosima.fastrtps.rtps.BuiltinAttributes.domainId` is a `uint32_t` and there is no restriction in place what values to set it to. So I am not sure if the rmw impl. should artificially limit the value range.

@richiprosima are all possible `uint32_t` values valid or has FastRTPS some restrictions on its range?

richiware, 2019-03-14:
Due to the DDS standard formula to calculate the ports, as @alsora said, there is a limitation. Currently we don't check the restriction, but we will. I will link the associated PR with this issue. Thanks.

richiware, 2019-04-12:
We are working on the next release and we want to introduce this limitation in this version.

MiguelCompany, 2019-06-10:
@alsora This should have been solved by eProsima/Fast-RTPS#496, which is part of 1.8.0 release. Can this issue be closed?

alsora, 2019-06-10:
@MiguelCompany Thank you, it works perfectly

closed 2019-06-10 by alsora