ros2/rmw_fastrtps #285: ABBA deadlock in rmw_fastrtps/Fast-RTPS

clalancette, 2019-06-03:
## Bug report

There seems to be a ABBA deadlock in how rmw_fastrtps and Fast-RTPS proper are taking locks.  This most easily shows up in a ROS 2 multi-threaded executor program.  I haven't been able to create a simple example that makes it happen, but in my complex example, it happens pretty regularly.  Note that I'm not sure if this is a bug in rmw_fastrtps or Fast-RTPS, but we'll start here.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - c8a0071e69bbcf8f3b45d060303aa806bd5962cb
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Run a complex MultiThreadedExecutor program.

#### Expected behavior

Program continues to run indefinitely.

#### Actual behavior

Program hangs after some amount of time.

#### Additional information

I tracked this down using gdb.  What's happening is a classic ABBA deadlock between rmw_fastrtps and Fast-RTPS.

In thread one, Fast-RTPS has received some data and is trying to deliver it.  While doing that, it takes the [StatefulReader::mp_mutex lock](https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/rtps/reader/StatefulReader.cpp#L224), and then continues on to call the previously registered rmw_fastrtps [onNewDataMessage callback](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L81).  `onNewDataMessage` tries to take [internalMutex_](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L149), but is blocked doing so.  The order of locks taken is Fast-RTPS mp_mutex followed by rmw_fastrtps internalMutex_.

In thread two, an executor wants to get take some already delivered data, so it calls [data_taken](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L135).  In turn, data_taken acquires the [internalMutex_](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L149), then calls down into Fast-RTPS [getUnreadCount](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp#L149) to get the data.  This tries to take [mp_mutex](https://github.com/eProsima/Fast-RTPS/blob/master/include/fastrtps/subscriber/SubscriberHistory.h#L102), but since this is the same mutex as was taken in thread 1, the whole thing deadlocks.  The order of locks taken is rmw_fastrtps internalMutex_ followed by Fast-RTPS mp_mutex.

@MiguelCompany @richiware I'm not entirely sure whether the problem is in Fast-RTPS itself, or in how rmw_fastrtps is interacting with it.  I will say that what is being protected by which locks in https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_subscriber_info.hpp is pretty unclear, so it may be a good idea to rethink and clean that up a bit.  Regardless, any thoughts here would be appreciated.

richiware, 2019-06-04:
I've noticed I introduced that deadlock on commit fc4b27. Trying to fix a race condition I didn't be aware of introducing a ABBA deadlock. Reading again the code of that commit I removed an internal variable that maybe is was the responsible of having the mutex there. I will take a look to your PR.

clalancette, 2019-06-04:
> I've noticed I introduced that deadlock on commit fc4b27. Trying to fix a race condition I didn't be aware of introducing a ABBA deadlock.

That's OK, I approved it and didn't notice the potential race either :).

> I will take a look to your PR.

Thanks, appreciated.

PR #286 (2019-06-04): Fix ABBA deadlock.
onNewDataMessage() is called as a callback from Fast-RTPS when
there is new data.  When that happens, some locks inside of
Fast-RTPS are taken, followed by internalMutex_ in SubListener.
However, the rest of SubListener takes internalMutex_ first,
followed by calls into Fast-RTPS where locks are taken.  This
is an ABBA deadlock.  Break the deadlock by doing the calls
into Fast-RTPS before taking the internalMutex_.

This should fix #285 

@MiguelCompany @richiprosima Any thoughts on this?
closed 2019-06-04 by clalancette