ros2/rmw_fastrtps #297: Messages stop 90s after network is disconnected (multicast is enabled on loopback)

clydemcqueen, 2019-06-27:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  binaries
- Version or commit hash:
  -  dashing
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  -  rclcpp

#### Steps to reproduce issue

 - boot with network enabled, e.g., Ethernet plugged in
 - enable multicast on loopback, e.g.,
```
route add -net 224.0.0.0 netmask 240.0.0.0 dev lo
ifconfig lo multicast
```
 - in terminal 1: `ros2 run demo_nodes_cpp talker`
 - in terminal 2: `ros2 run demo_nodes_cpp listener`
 - drop the network, e.g., unplug Ethernet

#### Expected behavior

Listener continues to report messages.

#### Actual behavior

Listener stops reporting messages after ~90 seconds.

#### Additional information

If the talker and listener are started while the network is down, then bringing the network up and down has no effect: the listener continues to report messages.

----
## Feature request

#### Feature description

#### Implementation considerations

jacobperron, 2019-08-22:
I can reproduce on 18.04 by enabling multicast with

    sudo ip route add 224.0.0.0/4 dev lo

Running the talker and listener and turning off my network interface.

I notice the listener stops logging messages about 90 seconds after the network is disabled and when I re-enable the network it continues logging messages after about 20 seconds.

I'm running the latest ROS 2 branches from source with Fast-RTPS.

jacobperron, 2019-08-22:
@richiware @MiguelCompany Any thoughts?

MiguelCompany, 2019-09-05:
@jacobperron I've been able to reproduce this using just a Fast-RTPS example. I've found the issue and a fix is on the way (eProsima/Fast-RTPS#695)

jacobperron, 2019-09-05:
> @jacobperron I've been able to reproduce this using just a Fast-RTPS example. I've found the issue and a fix is on the way (eProsima/Fast-RTPS#695)

Awesome, thanks!

MiguelCompany, 2019-09-06:
@jacobperron The PR has been merged into master. Would appreciate if you check on your side.

jacobperron, 2019-09-06:
The issue appears to be resolved for me using https://github.com/eProsima/Fast-RTPS/commit/acbc07b30742be65fac349e1eb9478b8aaa36c32.

Cheers!

closed 2019-09-06 by jacobperron