ros2/rmw_fastrtps #299: Thread safety warnings in rmw_node_info_and_types.cpp

emersonknapp, 2019-07-03:
## Bug report

The nightly thread safety build is marked unstable because some potentially thread-unsafe code was introduced -

> reading variable 'discovered_namespaces' requires holding mutex 'impl->listener->names_mutex_' [-Wthread-safety-analysis]

see the CI job for warning details https://ci.ros2.org/view/nightly/job/nightly_linux_clang_libcxx/67/warnings23Result/

**Required Info:**

- Operating System:
  - Linux x86_64
- Installation type:
  - CI Build
- Version or commit hash:
  - Last 36 builds
- DDS implementation:
  - rmw_fastrtps_shared_cpp
- Client library (if applicable):
  - N/A

ivanpauno, 2019-07-04:
I guess the warning was introduced in https://github.com/ros2/rmw_fastrtps/pull/283.

MiguelCompany, 2019-07-04:
> I guess the warning was introduced in #283.

Good guess. That PR introduced a `RCPPUTILS_TSA_GUARDED_BY` annotation for the two maps the warnings are complaining about. The change was necessary because tsan was complaining on concurrent access to both of them.

PR #306 (2019-07-29): Add missing lock guards for discovered_names and discovered_namespaces
Backport #301 
Fixes #299
closed 2019-07-09 by ivanpauno