ros2/rmw_fastrtps #313: Publisher can crash after too many unclean subscriber shutdowns

clalancette, 2019-09-04:
## Bug report

I've run into a scenario where a crashing subscriber can (eventually) cause a Fast-RTPS publisher to crash.  I have one node that is publishing a `sensor_msgs/msg/Imu` message on `/imu` (it is the 
code for https://github.com/KumarRobotics/imu_vn_100/tree/dashing , but I suspect anything will work here).  I have a second node which subscribes to the `/imu' topic, but intentionally crashes after 5 seconds: https://github.com/clalancette/ros2-subscriber-crasher .  I then run the publisher in one terminal, and the subscriber in a loop in a second terminal:

```
while true ; do ros2 run crasher crasher_node ; done
```

What I expect to happen is to see ~500 callbacks printed out from the subscriber, followed by a crash, followed by the subscriber starting up again, indefinitely.  What I actually see is that the first few times the crasher runs, it does indeed get ~500 callbacks.  After a while, the number of callbacks it is processing steadily drops until it gets to around 40, at which point the *publisher* crashes.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - Latest master as of this writing, 099f9eed9a0f581447405fbd877c6d3b15f1f26e
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Terminal one:
```
ros2 run imu_vn_100 imu_vn_100_node
```

Terminal two:
```
while true ; do ros2 run crasher crasher_node ; done
```

#### Expected behavior

Publisher and subscriber run forever, with the subscriber printing out ~500 callbacks every time it runs.

#### Actual behavior

Runs of the subscriber after the first few process fewer and fewer callbacks, and eventually crash the publisher.

#### Additional information
This example runs much longer (at least 15 minutes) when using CycloneDDS.

@MiguelCompany any thoughts on what may be happening here?

clalancette, 2019-09-06:
> > After a while, the number of callbacks it is processing steadily drops until it gets to around 40, at which point the _publisher_ crashes.
> 
> What does 'after a while' mean? 1 minute? 10 seconds?

Good question.  It varies run-to-run, but it is usually between 1 and 2 minutes.  I now am also noticing that as this is happening, the CPU usage of the publisher is steadily climbing until it is taking a 100% of a CPU.

> Do you have a core dump of the publisher crashing?

I don't currently.  I'll rebuild and try to grab one.

> 
> Looking at the code, I think the default QoS values are being used. That means KEEP_LAST 10, RELIABLE, VOLATILE both in publisher and subscriber. Can you confirm this is true?

Yes, I'm using default QoS values for both publisher and subscriber, so the values you have above are correct.

> 
> On which platforms are the publisher and subscriber running?

Both on Ubuntu Bionic, both running on the same machine.

Here's an example publisher that I've now written: https://github.com/clalancette/ros2-fake-imu-publisher .  You should be able to reproduce the problem by building and running the publisher and the subscriber in a ROS 2 workspace, then running in one terminal:

```
ros2 run fake_imu fake_imu_node
```

and in a second terminal

```
while true ; do ros2 run crasher crasher_node ; done
```

Let me know if you need any more information.

clalancette, 2019-09-06:
Here's the stack trace from the core dump:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007fa3a6beb801 in __GI_abort () at abort.c:79
#2  0x00007fa3a7240957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fa3a7246ab6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fa3a7246af1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fa3a7246aaa in std::rethrow_exception(std::__exception_ptr::exception_ptr) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fa3a7f3c953 in rclcpp::exceptions::throw_from_rcl_error (ret=1, prefix="failed to publish message", 
    error_state=0x0, reset_error=0x7fa3a774ff8b <rcutils_reset_error>)
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/exceptions.cpp:87
#7  0x000055a398eca5a5 in rclcpp::Publisher<sensor_msgs::msg::Imu_<std::allocator<void> >, std::allocator<void> >::do_inter_process_publish (this=0x55a39a3a5620, msg=0x55a39a41eca0)
    at /home/ubuntu/ros2_ws/install/rclcpp/include/rclcpp/publisher.hpp:222
#8  0x000055a398ec90fd in rclcpp::Publisher<sensor_msgs::msg::Imu_<std::allocator<void> >, std::allocator<void> >::publish (this=0x55a39a3a5620, msg=std::unique_ptr<sensor_msgs::msg::Imu_<std::allocator<void> >> = {...})
    at /home/ubuntu/ros2_ws/install/rclcpp/include/rclcpp/publisher.hpp:103
#9  0x000055a398ec847d in FakeImu::timer_cb (this=0x55a399f56930)
    at /home/ubuntu/crasher_ws/src/ros2-fake-imu-publisher/src/fake_imu_node.cpp:31
#10 0x000055a398ed755b in std::__invoke_impl<void, void (FakeImu::*&)(), FakeImu*&> (
    __f=@0x55a39a1ab4b8: (void (FakeImu::*)(FakeImu * const)) 0x55a398ec83c2 <FakeImu::timer_cb()>, 
    __t=@0x55a39a1ab4c8: 0x55a399f56930) at /usr/include/c++/7/bits/invoke.h:73
#11 0x000055a398ed73aa in std::__invoke<void (FakeImu::*&)(), FakeImu*&> (
    __fn=@0x55a39a1ab4b8: (void (FakeImu::*)(FakeImu * const)) 0x55a398ec83c2 <FakeImu::timer_cb()>, 
    __args#0=@0x55a39a1ab4c8: 0x55a399f56930) at /usr/include/c++/7/bits/invoke.h:95
#12 0x000055a398ed719b in std::_Bind<void (FakeImu::*(FakeImu*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>) (this=0x55a39a1ab4b8, __args=...) at /usr/include/c++/7/functional:467
#13 0x000055a398ed6c57 in std::_Bind<void (FakeImu::*(FakeImu*))()>::operator()<, void>() (this=0x55a39a1ab4b8)
    at /usr/include/c++/7/functional:551
#14 0x000055a398ed684a in rclcpp::GenericTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0>::execute_callback_delegate<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0>() (this=0x55a39a1ab490)
    at /home/ubuntu/ros2_ws/install/rclcpp/include/rclcpp/timer.hpp:168
#15 0x000055a398ed60cc in rclcpp::GenericTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0>::execute_callback() (this=0x55a39a1ab490) at /home/ubuntu/ros2_ws/install/rclcpp/include/rclcpp/timer.hpp:155
#16 0x00007fa3a7f40563 in rclcpp::executor::Executor::execute_timer (timer=warning: RTTI symbol not found for class 'std::_Sp_counted_ptr_inplace<rclcpp::WallTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0>, std::allocator<rclcpp::WallTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0> >, (__gnu_cxx::_Lock_policy)2>'
warning: RTTI symbol not found for class 'std::_Sp_counted_ptr_inplace<rclcpp::WallTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0>, std::allocator<rclcpp::WallTimer<std::_Bind<void (FakeImu::*(FakeImu*))()>, (void*)0> >, (__gnu_cxx::_Lock_policy)2>'

std::shared_ptr<rclcpp::TimerBase> (use count 3, weak count 1) = {...})
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:380
#17 0x00007fa3a7f3fa07 in rclcpp::executor::Executor::execute_any_executable (this=0x7ffdd987e550, any_exec=...)
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:285
#18 0x00007fa3a7f4b08a in rclcpp::executors::SingleThreadedExecutor::spin (this=0x7ffdd987e550)
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/single_threaded_executor.cpp:36
#19 0x00007fa3a7f457b8 in rclcpp::spin (node_ptr=
    std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface> (use count 6, weak count 1) = {...})
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executors.cpp:35
#20 0x00007fa3a7f458d6 in rclcpp::spin (node_ptr=std::shared_ptr<rclcpp::Node> (use count 2, weak count 1) = {...})
    at /home/ubuntu/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executors.cpp:42
#21 0x000055a398ec7695 in main (argc=1, argv=0x7ffdd987e878)
    at /home/ubuntu/crasher_ws/src/ros2-fake-imu-publisher/src/fake_imu_node.cpp:49
```

Unfortunately it isn't super useful since the crash happens because of an unhandled exception that was thrown.  The real error is suggested more by the error message we see on the console right before the crash:

```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/ubuntu/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/ubuntu/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:257
```

dirk-thomas, 2019-09-06:
This looks like a duplicate of ros2/ros2#714.

clalancette, 2019-09-10:
> This looks like a duplicate of [ros2/ros2#714](https://github.com/ros2/ros2/issues/714).

It does indeed look like that is the case.  I'll add a comment to that issue with the additional reproducer from this issue, and close this one.

closed 2019-09-10 by clalancette

ivanpauno, 2019-09-11:
https://github.com/ros2/ros2/issues/714 is a collection of examples where the publisher crashes, and in all of them the root cause seems to be a failure in `FastRTPS`.
But I'm not to sure if the failures are all of the same kind, or if they are different problems (we're basically catching all the exceptions and then publishing `cannot publish data`, which is not much informative).