ros2/rmw_fastrtps #317: Exception sending message over network

glmnet, 2019-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 
- DDS implementation:
  - fast rtps? I'm not sure of having chosen one
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I'm publishing odometry data from a robot, it is a differential drive one, using mostly turtlebot3 as example, many messages get published then randomly after some seconds between 5 to more than 200 may be it will throw an exception.

#### Expected behavior
No exceptions!

#### Actual behavior

The communication is throwing an exception

terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/rosbert/ros2_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/rosbert/ros2_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:257

#### Additional information
````
48:  rmw_fastrtps_shared_cpp::SerializedData data;
49:  data.is_cdr_buffer = false;
50:  data.data = const_cast<void *>(ros_message);
51:  if (!info->publisher_->write(&data)) {
52:    RMW_SET_ERROR_MSG("cannot publish data");
53:    return RMW_RET_ERROR;
54:  }

....

    rcl_ret_t
    rcl_publish(
      const rcl_publisher_t * publisher,
      const void * ros_message,
      rmw_publisher_allocation_t * allocation)
    {
      if (!rcl_publisher_is_valid(publisher)) {
        return RCL_RET_PUBLISHER_INVALID;  // error already set
      }
      RCL_CHECK_ARGUMENT_FOR_NULL(ros_message, RCL_RET_INVALID_ARGUMENT);
      if (rmw_publish(publisher->impl->rmw_handle, ros_message, allocation) != RMW_RET_OK) {
257:    RCL_SET_ERROR_MSG(rmw_get_error_string().str);
        return RCL_RET_ERROR;
      }
      return RCL_RET_OK;
    }

````

richiware, 2019-09-16:
I was investigating using the suggested commands:

```bash
ros2 run image_tools cam2image -b -x 2000 -y 9000
ros2 run image_tools showimage
```

As I said in a previous comment, publisher's `write()` function is returning `false` because the `max_blocking_time` is exceeded. FastRTPS v1.9 is more real-time aware and any `write()` operation will never exceed `max_blocking_time`. By default it is 100ms.

This example is sending samples which are larger than 45Mb, and the publisher's history depth is 10. Also the video rate is high. It is a lot of data to be send by the asynchronous thread, and sometimes it takes more than 100ms blocking the user's thread.

I was working to improve the asynchronous thread publication. But I have to automatize the calculation of results to draw conclusions. Until now I was using my perception seeing the output. I will create a script to show real rates and other information.

Anyway the user's thread could fail with so amount of data. A workaround could be increase the `max_blocking_time` in the _rmw_fastrtps_.

Also I was testing opensplice. I don't have real rates until I create the script, only my perception, but I think opensplice uses the user's thread to send the sample, blocking for as long as it takes. This causes the decreasing of the video rate. It sends the samples slower because it is blocking the user's thread.

@clalancette About your example which forces a crash, the explanation is similar. Until the publisher detects the crash warned by the liveliness mechanism (default lease duration is 130 seconds), the publisher sends a sample to each subscriber's unicast port. The more subscribers the longer the asynchronous thread is blocking the user thread. It is probably taking more than 100ms.

quhezheng, 2019-09-16:
> I was investigating using the suggested commands:
> 
> ```shell
> ros2 run image_tools cam2image -b -x 2000 -y 9000
> ros2 run image_tools showimage
> ```
> 
> As I said in a previous comment, publisher's `write()` function is returning `false` because the `max_blocking_time` is exceeded. FastRTPS v1.9 is more real-time aware and any `write()` operation will never exceed `max_blocking_time`. By default it is 100ms.
> 
> This example is sending samples which are larger than 45Mb, and the publisher's history depth is 10. Also the video rate is high. It is a lot of data to be send by the asynchronous thread, and sometimes it takes more than 100ms blocking the user's thread.
> 
> I was working to improve the asynchronous thread publication. But I have to automatize the calculation of results to draw conclusions. Until now I was using my perception seeing the output. I will create a script to show real rates and other information.
> 
> Anyway the user's thread could fail with so amount of data. A workaround could be increase the `max_blocking_time` in the _rmw_fastrtps_.
> 
> Also I was testing opensplice. I don't have real rates until I create the script, only my perception, but I think opensplice uses the user's thread to send the sample, blocking for as long as it takes. This causes the decreasing of the video rate. It sends the samples slower because it is blocking the user's thread.
> 
> @clalancette About your example which forces a crash, the explanation is similar. Until the publisher detects the crash warned by the liveliness mechanism (default lease duration is 130 seconds), the publisher sends a sample to each subscriber's unicast port. The more subscribers the longer the asynchronous thread is blocking the user thread. It is probably taking more than 100ms.

@richiware How to configure the `max_blocking_time`  in the rmw_fastrtps ? The only place where setup the default value is constructor of ReliabilityQosPolicy:
``` shell
          // max_blocking_time = 100ms  
          max_blocking_time{0, 4294967100} (why the magic number?)
```
Any sample code to tell us how to workaround this bug? It's killing our team members

richiware, 2019-09-16:
@quhezheng _rmw_fastrtps_ is able to load a default XML file containing QoS and use them. Write an XML file named `DEFAULT_FASTRTPS_PROFILES.xml` in the working directory (directory where you execute the command) with this content:

```xml
<?xml version="1.0" encoding="UTF-8" ?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
    <profiles>
        <publisher profile_name="publisher_increase_max_blocking_time" is_default_profile="true">
            <qos>
                <reliability>
                    <kind>RELIABLE</kind>
                    <max_blocking_time>
                        <sec>1</sec>
                        <nanosec>0</nanosec>
                    </max_blocking_time>
                </reliability>
            </qos>
        </publisher>
    </profiles>
</dds>
```

ivanpauno, 2019-09-16:
Thanks for investigating the issue @richiware !
The reliability `QoS` policy of both nodes I commented (`showimage` and `cam2image`) is best effort.
Should `max_blocking_time` time affect them?

richiware, 2019-09-17:
Yes, current version 1.9.0 uses reliability QoS `max_blocking_time` to maintain strict real-time, both reliable and besteffort. Sure not all users want this feature. We met and decided to make this configurable. This will be released in v1.9.1 via eProsima/Fast-RTPS#718. If not configured strict real-time, the behaviour of `max_blocking_time` will be the defined by DDS standard.

ivanpauno, 2019-09-19:
> Yes, current version 1.9.0 uses reliability QoS `max_blocking_time` to maintain strict real-time, both reliable and besteffort. Sure not all users want this feature. We met and decided to make this [...]

Thanks for the answer and the patch!
I think it's good to stick to the DDS standard. Also, having an extra option for people that want strict real time sounds nice for both reliable and best-effort sounds nice.

I also see that we have to do some changes in our side, in order to use reliable communications properly (e.g.: expose `max_blocking_time`).

extra question:
Is there a way to know if an operation failed because the blocking time was exceeded?
e.g.: I see that `write` method just returns a bool, and I don't know if the error type is available somehow.

quhezheng, 2019-09-20:
> @quhezheng _rmw_fastrtps_ is able to load a default XML file containing QoS and use them. Write an XML file named `DEFAULT_FASTRTPS_PROFILES.xml` in the working directory (directory where you execute the command) with this content:
> 
> ```
> <?xml version="1.0" encoding="UTF-8" ?>
> <dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles" >
>     <profiles>
>         <publisher profile_name="publisher_increase_max_blocking_time" is_default_profile="true">
>             <qos>
>                 <reliability>
>                     <kind>RELIABLE</kind>
>                     <max_blocking_time>
>                         <sec>1</sec>
>                         <nanosec>0</nanosec>
>                     </max_blocking_time>
>                 </reliability>
>             </qos>
>         </publisher>
>     </profiles>
> </dds>
> ```

@richiware Many thanks. 
I wend to back to 1.7.0, there was no locking op. I don't understand what the lock it protecting? Can we remove the lock OP for short term in our local code base? What's consequence would it bring if remove it??

richiware, 2019-09-20:
@ivanpauno Currently there is no way to know if writing fails because of a timeout. It is still old API. We are working on have the standard DDS PSM C++ API ([working branch](https://github.com/eProsima/Fast-RTPS/tree/feature/pubsub_layer_changes)). This API contemplates that case.

@quhezheng  What version of FastRTPS are you using? v1.8.x? We can make a back port of eProsima/Fast-RTPS#718 to v1.8.x.

quhezheng, 2019-09-21:
> @ivanpauno Currently there is no way to know if writing fails because of a timeout. It is still old API. We are working on have the standard DDS PSM C++ API ([working [...]

@richiware we are using v 1.8.1，should we -DSTRICT_REALTIME= OFF to workaround our case?

richiware, 2019-09-23:
> @richiware we are using v 1.8.1，should we -DSTRICT_REALTIME= OFF to workaround our case?

Until now the solution was able only in 1.9.x. I've created eProsima/Fast-RTPS#730 with the backport to 1.8.x. By default STRICT_REALTIME will be OFF. I hope this helps you.

tompe17, 2019-10-02:
I am running the binary version of dashing and have this problem when using the parameter bridge to send messages between four ROS1 agents on different computers each with its own ROS core.  Setting the timeout to 10s eliminated the crashes but I do not  get fairness in the communication. I get chunks of data from one computer.

I have two questions. Is this a bug that will be resolved if I wait?

And is there a configuration to get fairness in the communication? (Maybe this is an unreleted issue..)

ivanpauno, 2019-10-02:
I tested both the `image_tools` example and @clalancette [example](https://github.com/ros2/rmw_fastrtps/issues/317#issuecomment-530078825) and they aren't crashing more (using current master, installation from source).

Could you @glmnet @realdream @Jconn @SteveMacenski @maxlein check if the problem is also solved in your respective examples?

> I have two questions. Is this a bug that will be resolved if I wait?

It's already solved if you are installing from source and using master, it may be backported to `Dashing` in next patch release.

> And is there a configuration to get fairness in the communication?

Sorry, I don't understand the question.

Jconn, 2020-02-13:
This stopped reproducing for me on dashing, and have not seen it on eloquent. I'm comfy closing this

ivanpauno, 2020-02-14:
> This stopped reproducing for me on dashing, and have not seen it on eloquent. I'm comfy closing this

Closing! It can be reopened if it's still a problem.

closed 2020-02-14 by ivanpauno

Tirthankar-Halder, 2024-08-20:
I am facing a similar issue. I am unable to figure out how to solve this issue.
I am using TI ROS driver packages from [Ti Ros Driver](git clone https://git.ti.com/git/mmwave_radar/mmwave_ti_ros.git
) instructed in [Installation documentation](https://dev.ti.com/tirex/explore/node?node=A__AFIEVaaBoBCgo.VCtJenZQ__radar_toolbox__1AslXXD__LATEST)

Operating System: Ubuntu 22.04
Ros  : Humble
Error :  [ParameterParser-4] cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'

```
[ParameterParser-4] cannot publish data, at ./src/rmw_publish.cpp:62 during '__function__'
[ParameterParser-4] destroy_client() failed to delete datareader, at ./src/rmw_client.cpp:92 during '__function__'
[ERROR] [ParameterParser-4]: process has died [pid 14306, exit code -11, cmd '/home/tirthapc/mmwave_ti_ros/ros2_driver/install/ti_mmwave_rospkg/lib/ti_mmwave_rospkg/ParameterParser --ros-args -r __node:=ParameterParser --params-file /tmp/launch_params_u4j6h_8v --params-file /tmp/launch_params_mi129we3 --params-file /tmp/launch_params_oi0w82_e'].
```
I get this output for "dpkg -l | grep fastrt"
[image]

MiguelCompany, 2025-06-23:
1. Remember that cross-distribution communication is not guaranteed to work, and unexpected issues may arise when mixing rolling with humble.
2. You should capture possible exceptions and act accordingly