ros2/rmw_fastrtps #319: Getting QoS policies of all the publishers publishing to and subscribers subscribing to a topic

jaisontj, 2019-09-19:
## Feature request

#### Feature description
As a part of implementing [this](https://github.com/ros2/rmw/issues/151) feature for the ros2cli, some form of map of topic to publisher and subscriber qos policies need to be maintained.

#### Implementation considerations
Similar to how [`rmw_count_publishers` and `rmw_count_subscribers`](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/rmw_count.cpp#L73) works; `custom_participant_info.hpp` has two variables; [`reader_topic_cache` and `writer_topic_cache`](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L190) which can be used to get a list of subscribers and publishers to a topic represented by their GUID. 

**Question 1: Is there a way to get Publisher and Subscriber attributes from their respective GUIDs?** 

Assuming that the answer to the above question is "No", I was thinking that a new `qos_cache.hpp` could be created which stores a map of `GUID` to `rmw_qos_profile_t` and this map could be populated inside the [`process_discovery_info`](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L163) method found inside `custom_participant_info.hpp` in a manner similar to how `topic_cache`.add/removeTopic works, i.e get the [`Writer/ReaderQos`](https://github.com/eProsima/Fast-RTPS/blob/master/src/cpp/qos/WriterQos.cpp) from their corresponding [`Writer/ReaderProxyData`](https://github.com/eProsima/Fast-RTPS/blob/master/include/fastrtps/rtps/builtin/data/WriterProxyData.h#L310).

This leaves us with figuring out a way to convert `Writer/ReaderQos` to `rmw_qos_profile_t`; There is a function [`dds_qos_to_rmw_qos`](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/src/qos.cpp#L168) inside `qos.hpp`. However, this method expects data of type `SubscriberAttributes` or `PublisherAttributes`; this is because `rmw_qos_profile` stores `history` and `depth` data which comes from `TopicAttributes`.

**Question 2: How do we go about getting the TopicAttributes in this case?** 

**Question 3: Should the `dds_qos_to_rmw_qos` function be using two functions inside: one to populate data from `TopicAttributes` and another to populate data from `Writer/ReaderQos` (this code can now be reused for our use-case)?**

**Question 4: Is there a better way to do what I'm trying to do?**

ivanpauno, 2019-09-26:
> Question 1: Is there a way to get Publisher and Subscriber attributes from their respective GUIDs?
>
> Assuming that the answer to the above question is "No", I was thinking that a new qos_cache.hpp [...]

AFAIK, you're analysis is correct. There is no way to access the data if you don't store it at discovery.

> Haven't check thoroughly, but if it's not part of the discovery data I don't see a way of recover it.

> Question 3: Should the dds_qos_to_rmw_qos function be using two functions inside: one to populate data from TopicAttributes and another to populate data from Writer/ReaderQos (this code can now be reused for our use-case)?

Sure! any change that improves code reuse is ok.

> Question 4: Is there a better way to do what I'm trying to do?

I would have done the same, so I don't have a better way.

ivanpauno, 2019-09-26:
> this is because rmw_qos_profile stores history and depth data which comes from TopicAttributes.

@richiprosima Is this somehow available from the publisher/subscriber discovery information?

MiguelCompany, 2019-09-27:
@ivanpauno Yes, the discovery info structure has an XXXProxyData info field, that is the one processed [here](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L163). This structure has an XXXQos m_qos field with the QoS parameters of the discovered endpoint. See [here](https://github.com/eProsima/Fast-RTPS/blob/master/include/fastrtps/rtps/builtin/data/WriterProxyData.h#L310) and [here](https://github.com/eProsima/Fast-RTPS/blob/master/include/fastrtps/rtps/builtin/data/ReaderProxyData.h#L309)

ivanpauno, 2019-09-27:
@richiprosima Sorry for the unclear question.
I see that `WritterQoS` is available in `WritterProxyData`, though some of our qos settings like `history_kind` and `history_depth` don't seem to be part of the `WritterQoS`, but of [HistoryQos](https://github.com/eProsima/Fast-RTPS/blob/9d562024886f4e1e7be363356ab032ecba934490/include/fastrtps/qos/QosPolicies.h#L759-L760) which is part of the [TopicAttributes](https://github.com/eProsima/Fast-RTPS/blob/9d562024886f4e1e7be363356ab032ecba934490/include/fastrtps/attributes/TopicAttributes.h#L115) that are available in [PublisherAttributes](https://github.com/eProsima/Fast-RTPS/blob/9d562024886f4e1e7be363356ab032ecba934490/include/fastrtps/attributes/PublisherAttributes.h#L64-L68) (and not directly in the `WritterQoS`). `HistoryQoS` nor `TopicAttributes` seems to be available in [WriterProxyData](https://github.com/eProsima/Fast-RTPS/blob/master/include/fastrtps/rtps/builtin/data/WriterProxyData.h).
I wonder if `HistoryQoS` policy is available somehow or not.

jaisontj, 2019-10-11:
@ivanpauno I want to display the participant name along with their QoS Profile. For this I plan on leveraging the [discovered_names](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L188) map, but I've observed that some participants do not have names and therefore do not have an entry in the `discovered_names` map. 

_**Question 1: Why is it that some participants do not have names?**_

_**Question 2: My plan is to display their name as `NAME_NOT_FOUND` and display their QoS Policy anyway. Does it make sense, will this have some unintended side effects?**_

_**Question 3: Is there another way to retrieve participant name that I may be missing?**_

ivanpauno, 2019-10-14:
> Question 1: Why is it that some participants do not have names?

All nodes have names. All participants associated with a node have names.
Some DDS vendors provide a participant name (IIRC, only `Fast-RTPS`).
If not, we use the `user data` qos policy to communicate the node name.

Still, the `discovered_names` map should have all the node names.
Participants created using non-ros code may not have a name (nor use our `user data` convention).

> Question 2: My plan is to display their name as NAME_NOT_FOUND and display their QoS Policy anyway. Does it make sense, will this have some unintended side effects?

IMO, If the participant doesn't have an entry in [discovered_names](https://github.com/ros2/rmw_fastrtps/blob/master/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L188), we could left the node name empty.
It should be understanded as a publisher/subscription created by non-ros code and still following ROS topic name conventions (or ROS code using non-demangle option).

> Question 4: Have I missed something with the above change in approach?

Using the same map sounds reasonable.
It should be something like:
`typedef std::unordered_map<std::string, TopicData> TopicToTypes;`

And `TopicData` should be an struct with the types and the qos policies.
Remember that you may find more than one `Publisher/Subscription` created by the same node and using the same topic name, so you may have many `qos policies` items in each `TopicData` (not only one).

jaisontj, 2019-10-14:
> All nodes have names. All participants associated with a node have names.
> Some DDS vendors provide a participant name (IIRC, only `Fast-RTPS`).
> If not, we use the `user data` qos policy to [...]

Alright!

> IMO, If the participant doesn't have an entry in [...]

If this is obvious then leaving it empty makes sense, but if not, leaving something that might help the user identify why some publishers/subscribers do not have names might be helpful IMO.

> Using the same map sounds reasonable.
> It should be something like:
> `typedef std::unordered_map<std::string, TopicData> TopicToTypes;`
>
> And `TopicData` should be an struct with the types and the qos [...]

Hmm, interesting. The reason why I went with including the `QoS Policy` in the `ParticipantTopicMap` was because 
- we are associating `QoS Policies` with each publisher's/subscriber's `GUID_t` 
- avoid duplication of the `QoS Policy` since publishers and subscribers can be associated with multiple topics. 
- In case we need to get the QoS Policy for a specific `GUID_t` in the future, as a separate feature.

Is there a reason why you prefer adding the `QoS Policy` to `TopicToTypes` instead, apart from the advantage of making retrieval of data easy.

Moreover, just to make sure that I've understood correctly, the `TopicData` structure would have a list of topic types and another list of a struct that encapsulates a Participant's name (gotten from `discovered_names`) and its respective QoS Profile (`rmw_qos_profile_t`) (like `rmw_participant_qos_profile_t` created in this [PR](https://github.com/ros2/rmw/pull/186/files/24a3908ec4d5c4ef520ddc0a8989f1f606a53640#diff-13f765047eb0e7fb997c3838c6130cadR264)). Is that correct?

ivanpauno, 2019-10-14:
> If this is obvious then leaving it empty makes sense, but if not, leaving something that might help the user identify why some publishers/subscribers do not have names might be helpful IMO.

Leaving a message is fine too.
Please, left something that is not a valid node name (see https://github.com/ros2/rmw/blob/ca572aa2de3082ace9b8572011c393630745e010/rmw/include/rmw/validate_node_name.h#L34-L40 and https://github.com/ros2/rmw/blob/ca572aa2de3082ace9b8572011c393630745e010/rmw/include/rmw/validate_namespace.h#L40-L42).

> we are associating QoS Policies with each publisher's/subscriber's GUID_t

Watch out, it's the participant GUID, not the publisher/subscriber GUID (see https://github.com/ros2/rmw_fastrtps/blob/455d2149f86b58fa2ca5ddae393cc352c0c1a13b/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L172-L173).
So, each entry to the map is directly associated with a node.

jaisontj, 2019-10-14:
> Watch out, it's the participant GUID, not the publisher/subscriber GUID (see
> 
> https://github.com/ros2/rmw_fastrtps/blob/455d2149f86b58fa2ca5ddae393cc352c0c1a13b/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L172-L173
> 
> ).
> So, each entry to the map is directly associated with a node (well, with a participant).

This is a little confusing because this line resides in a function which is called from `onSubscriberDiscovery` and `onPublisherDiscovery`. Therefore, I assumed that the GUIDs would correspond to `Subscriber` and `Publisher` GUIDs respectively. 

Ok, so to summarize on the design:
- Create a new struct `TopicData` 

```c++
struct TopicData {
  vector<string> types;
  vector<rmw_qos_profile_t> qos_profiles;
}
```
Here, `types` and `qos_profiles` have the same indexing.

- update `typedef std::unordered_map<std::string, TopicData> TopicToTypes;`
**_Question 1: Should `ParticipantToTopicMap` also be updated to have `TopicData`?_**

- update `TopicCache.addTopic` to 

```c++
bool addTopic(
    const eprosima::fastrtps::rtps::InstanceHandle_t & rtpsParticipantKey,
    const std::string & topic_name,
    const std::string & type_name,
    const WriterQos & dds_qos) {..}
```
**_Question 2: Should `addTopic` function be overloaded for `WriterQos` and `ReaderQos` or just use a template variable?_**

**_- Question 3: how will `removeTopic` work. Are `topic_types` (or at least the combination of `topic_type`, `topic_name` and `qos_profile`) unique ? Otherwise I am not sure how I can identify the right QoS Policy to remove._**

ivanpauno, 2019-10-15:
> Question 1: Should ParticipantToTopicMap also be updated to have TopicData?

If we only need the "by topic" info, and we don't provide an equivalent "by node" introspection ability, `ParticipantToTopicMap` doesn't need to be updated.

> Question 2: Should addTopic function be overloaded for WriterQos and ReaderQos or just use a template variable?

Both a template or taking directly `rmw_qos_profile_t` sounds reasonable.

> Question 3: how will removeTopic work. Are topic_types (or at least the combination of topic_type, topic_name and qos_profile) unique ? Otherwise I am not sure how I can identify the right QoS Policy to remove.

Ha, good question.
What if we add the publisher/subscription GUID to the TopicData struct?
That will provide an unique way of identifying each publisher/subscription.

jaisontj, 2019-10-15:
Thank you for answering these @ivanpauno 

> > Question 3: how will removeTopic work. Are topic_types (or at least the combination of topic_type, topic_name and qos_profile) unique ? Otherwise I am not sure how I can identify the right QoS Policy to remove.
> 
> Good question.
> What if we add the publisher/subscription GUID to the TopicData struct?
> That will provide an unique way of identifying each publisher/subscription.

Yeah, that seems to be the only way. This is why I initially thought that modifying the `ParticipantQosMap

So here is what I am thinking about the struct:

```c++
struct TopicData {
  vector<GUID_t> guids;
  vector<string> types;
  vector<rmw_qos_profile_t> qos_profiles;
}
```
where all three have the same indexing.

Alternatively, we could also have them be a `vector<tuple<guid_t, string, rmw_qos_profile_t>>`

Is there a reason to go for separate arrays with same indexing as opposed to encapsulating it into one type? 

Let me know what you think?

ivanpauno, 2019-10-15:
> Is there a reason to go for separate arrays with same indexing as opposed to encapsulating it into one type?

Encapsulating in only one type sounds reasonable, whatever you prefer.

emersonknapp, 2020-03-04:
@dirk-thomas this one is also complete

closed 2020-03-05 by dirk-thomas