ros2/rmw_fastrtps #321: graph guard condition not triggered when new nodes are discovered

wjwwood, 2019-09-20:
Looking at this function which adds/removes node information based on discovery events, it never triggers the node graph guard condition(s):

https://github.com/ros2/rmw_fastrtps/blob/2017684f26dce6b85166ccd8df253eabc94acf81/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L60-L116

This makes it impossible to wait for a node to become available, because you cannot know when the answer to `get_node_names()` changes.

Therefore it should trigger this "graph" guard condition so it is possible to wait and not poll the node names.

jaisontj, 2019-10-03:
I am assuming that the fix will be similar to [this](https://github.com/ros2/rmw_fastrtps/blob/2017684f26dce6b85166ccd8df253eabc94acf81/rmw_fastrtps_shared_cpp/include/rmw_fastrtps_shared_cpp/custom_participant_info.hpp#L179-L183). Basically, if a new name is discovered and added successfully, then trigger the guard?

clalancette, 2023-06-13:
This is actually triggered nowadays via `add_participant` and `remove_participant`, as far as I can tell.  So closing this out.

closed 2023-06-13 by clalancette