ros2/rmw_fastrtps #324: Realsense D435 subscribing to raw camera from different computer throws exception

Jconn, 2019-09-25:
- Operating System:
Ubuntu 18.04
- Installation type:
binary
- Version or commit hash:
  from ros dashing binary 0.7.5 
- DDS implementation:
fast rtps
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue
Use librealsense [ros2 package ](https://github.com/intel/ros2_intel_realsense)to publish image data from the realsense D435. 

Subscribe to image data from a seperate computer. 

Observe exceptions being printed out from the realsense node with output:

`(source.cpp:127) Exception was thrown during user callback!
`

```

#### Expected behavior
no error output
#### Actual behavior
some error messages that don't inhibit behavior

#### Additional information
I still received messages from the realsense publisher even when I was getting some exception messages printed out.

I am still not sure what is causing this, but the issue stopped reproducing when I switched to opensplice, thus my fastRTPS suspicion.

mabelzhang, 2019-10-10:
Thank you for the report. Could you please provide a backtrace?

gonzodepedro, 2019-10-16:
A backtrace would be very useful to diagnose the issue.

Jconn, 2019-10-17:
I will get that backtrace tomorrow. I thought I had one logged somewhere but I can't find it

Jconn, 2019-10-18:
Okay I finally got that backtrace. I'm not sure if this is a fastrtps issue, I just know that I don't get this issue reproducing with opensplice community edition. I haven't tried other dds layers. 

steps to reproduce:

- On computer 1 launch realsense node with gdb, with 640x480 resolution, 30 depth fps, 6 rgb fps
- On computer 2 run `ros2 topic echo /camera/color/image_raw`

this debug information is printed out:

`(source.cpp:127) Exception was thrown during user callback!`

The backtrace through gdb after issuing `catch catch`:

```
#0  0x00007ffff5585b8f in __cxa_begin_catch () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff74b7d9b in librealsense::frame_source::invoke_callback(librealsense::frame_holder) const
    () from /opt/ros/dashing/lib/librealsense2.so.2
#2  0x00007ffff73a8c53 in librealsense::synthetic_source::frame_ready(librealsense::frame_holder) ()
   from /opt/ros/dashing/lib/librealsense2.so.2
#3  0x00007ffff73b34fe in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#4  0x00007ffff73b3868 in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#5  0x00007ffff73a9344 in librealsense::processing_block::invoke(librealsense::frame_holder) ()
   from /opt/ros/dashing/lib/librealsense2.so.2
#6  0x00007ffff748cee6 in rs2_process_frame () from /opt/ros/dashing/lib/librealsense2.so.2
#7  0x0000555555573318 in rs2::processing_block::invoke (f=..., this=<optimized out>)
    at /opt/ros/dashing/include/librealsense2/hpp/rs_processing.hpp:242
#8  realsense_ros2_camera::PipelineSyncer::operator() (f=..., this=<optimized out>)
    at /home/mekaworks2/grabber_ws/src/ros2_intel_realsense/realsense_ros2_camera/src/realsense_camera_node.cpp:95
#9  rs2::frame_callback<realsense_ros2_camera::PipelineSyncer>::on_frame (this=<optimized out>, 
    fref=<optimized out>) at /opt/ros/dashing/include/librealsense2/hpp/rs_sensor.hpp:109
#10 0x00007ffff74b7bd0 in librealsense::frame_source::invoke_callback(librealsense::frame_holder) const
    () from /opt/ros/dashing/lib/librealsense2.so.2
#11 0x00007ffff74508ab in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#12 0x00007ffff7451437 in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#13 0x00007ffff72ed9a4 in librealsense::platform::v4l_uvc_device::poll() ()
   from /opt/ros/dashing/lib/librealsense2.so.2
#14 0x00007ffff72ee118 in librealsense::platform::v4l_uvc_device::capture_loop() ()
   from /opt/ros/dashing/lib/librealsense2.so.2
---Type <return> to continue, or q <return> to quit---
#15 0x00007ffff55b166f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#16 0x00007ffff37756db in start_thread (arg=0x7fffc3fff700) at pthread_create.c:463
#17 0x00007ffff4c6e88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) bt
#0  0x00007ffff5585b8f in __cxa_begin_catch () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007ffff74b7d9b in librealsense::frame_source::invoke_callback(librealsense::frame_holder) const () from /opt/ros/dashing/lib/librealsense2.so.2
#2  0x00007ffff73a8c53 in librealsense::synthetic_source::frame_ready(librealsense::frame_holder) () from /opt/ros/dashing/lib/librealsense2.so.2
#3  0x00007ffff73b34fe in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#4  0x00007ffff73b3868 in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#5  0x00007ffff73a9344 in librealsense::processing_block::invoke(librealsense::frame_holder) () from /opt/ros/dashing/lib/librealsense2.so.2
#6  0x00007ffff748cee6 in rs2_process_frame () from /opt/ros/dashing/lib/librealsense2.so.2
#7  0x0000555555573318 in rs2::processing_block::invoke (f=..., this=<optimized out>) at /opt/ros/dashing/include/librealsense2/hpp/rs_processing.hpp:242
#8  realsense_ros2_camera::PipelineSyncer::operator() (f=..., this=<optimized out>) at /home/mekaworks2/grabber_ws/src/ros2_intel_realsense/realsense_ros2_camera/src/realsense_camera_node.cpp:95
#9  rs2::frame_callback<realsense_ros2_camera::PipelineSyncer>::on_frame (this=<optimized out>, fref=<optimized out>) at /opt/ros/dashing/include/librealsense2/hpp/rs_sensor.hpp:109
#10 0x00007ffff74b7bd0 in librealsense::frame_source::invoke_callback(librealsense::frame_holder) const () from /opt/ros/dashing/lib/librealsense2.so.2
#11 0x00007ffff74508ab in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#12 0x00007ffff7451437 in ?? () from /opt/ros/dashing/lib/librealsense2.so.2
#13 0x00007ffff72ed9a4 in librealsense::platform::v4l_uvc_device::poll() () from /opt/ros/dashing/lib/librealsense2.so.2
#14 0x00007ffff72ee118 in librealsense::platform::v4l_uvc_device::capture_loop() () from /opt/ros/dashing/lib/librealsense2.so.2
#15 0x00007ffff55b166f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#16 0x00007ffff37756db in start_thread (arg=0x7fffc3fff700) at pthread_create.c:463
#17 0x00007ffff4c6e88f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

jacobperron, 2019-10-22:
Thanks for the backtrace.

> (source.cpp:127) Exception was thrown during user callback!

Pretty sure this is where the error message is coming from:

https://github.com/IntelRealSense/librealsense/blob/31ea4aab7348923275a2a347bcaec7dba1b2bf5e/src/source.cpp#L127

Too bad it's not logging more details about the exception...

My best guess is that this callback function is throwing:

https://github.com/intel/ros2_intel_realsense/blob/8bc9a3de955bd0b6c2798728d444f4b9004a21fc/realsense_ros2_camera/src/realsense_camera_node.cpp#L496

Maybe you could try building the ROS driver from source and modifying that callback to try-catch and print any exceptions that occur, e.g.

```c++
auto frame_callback = [this](rs2::frame frame)
{
  try
  {
    // implementation ... 
  }
  catch(const std::runtime_error & ex)
  {
    RCLCPP_ERROR(logger_, "Error occurred in frame callback: %s", ex.what());
  }
};
```

Jconn, 2020-02-13:
With the changes to changes to the realsense2 node and moving to eloquent, I'm not hitting this anymore. Thanks for the help, I am closing

closed 2020-02-13 by Jconn