ros2/rmw_fastrtps #325: Question: Is there way to change DDS protocol setting UDP to TCP?

SlimyCoffee, 2019-09-27:
Can I post a question here? If not, I'm sorry and I'll delete this post right away.

I'm trying to combine new ROS2 system to existing one, which uses TCP only for some reason.

If ROS2 DDS implementation supports TCP, the combining work will get easy.

I've been looking for a long time, but I haven't found any information, about "how to change DDS setting" or like that. And that's why I'm asking questions here.

clalancette, 2019-09-27:
A better place to ask questions is on https://answers.ros.org; this consolidates all of our questions together in one place and makes it more searchable.  I'll close this out for now, but please feel free to follow-up there.

closed 2019-09-27 by clalancette

SlimyCoffee, 2019-09-27:
I got it. I'm sorry about your inconvenience.

Actually, I already posted the same question ROS answers, before posting it
to git issue.

https://answers.ros.org/question/333982/is-there-any-way-to-use-tcpip-in-ros2-dds-communication/

if you don't mind, could you visit this link and leave some comments?