ros2/rmw_fastrtps #337: Liveliness assertion errors are a bit loud

hidmic, 2019-11-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04, both on AMD64 and ARM64
- Installation type:
  - Binaries and from source
- Version or commit hash:
  - https://github.com/ros2/rmw_fastrtps/commit/a16c45ef11a19361a15c585497bf373a7ed0bae6 (https://github.com/eProsima/Fast-RTPS/commit/141b778c66c4f1f7aeb5c55551c173e3237beae9 on `1.9.x` at [eProsima/Fast-RTPS](https://github.com/eProsima/Fast-RTPS))
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Run one of the liveliness QoS demos e.g.:

```sh
ros2 run quality_of_service_demo_cpp liveliness 250 --node-assert-period 0 --policy MANUAL_BY_NODE
```

#### Expected behavior
Normal demo output:

```bash
$ ros2 run quality_of_service_demo_cpp liveliness 250 --node-assert-period 0 --policy MANUAL_BY_NODE
[INFO] [talker]: Talker starting up
[INFO] [listener]: Listener starting up
[INFO] [talker]: Publishing: 'Talker says 0'
[INFO] [listener]: Listener heard: [Talker says 0]
Liveliness changed event: 
  alive_count: 1
  not_alive_count: 0
  alive_count_change: 1
  not_alive_count_change: 0
Liveliness changed event: 
  alive_count: 0
  not_alive_count: 1
  alive_count_change: -1
  not_alive_count_change: 1
```
#### Actual behavior
Fast-RTPS error logs come along too:
 
```bash
$ ros2 run quality_of_service_demo_cpp liveliness 250 --node-assert-period 0 --policy MANUAL_BY_NODE
[INFO] [talker]: Talker starting up
[INFO] [listener]: Listener starting up
[INFO] [talker]: Publishing: 'Talker says 0'
[INFO] [listener]: Listener heard: [Talker says 0]
2019-11-13 20:37:26.670 [RTPS_WRITER Error] Error when restarting liveliness timer: 1 writers, liveliness 0 -> Function assert_liveliness
Liveliness changed event: 
  alive_count: 1
  not_alive_count: 0
  alive_count_change: 1
  not_alive_count_change: 0
Liveliness changed event: 
  alive_count: 0
  not_alive_count: 1
  alive_count_change: -1
  not_alive_count_change: 1
```
#### Additional information

Logs are coming from [this place](https://github.com/eProsima/Fast-RTPS/blob/c927b8555dfc87c3924d703cd354cf760d76ad21/src/cpp/rtps/writer/LivelinessManager.cpp#L209-L216).

ivanpauno, 2019-11-18:
@hidmic It seems to be already addressed in Fast-RTPS, can we close this?

hidmic, 2019-11-18:
Indeed, thanks @richiware !

closed 2019-11-18 by hidmic