ros2/rmw_fastrtps #341: Crashing when unsubscribing and resubscribing

SteveMacenski, 2019-12-10:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type: Shipped with Dashing & master both 
  - 
- Version or commit hash:
  - 
- DDS implementation: Fast-RTPS 
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

Able to unsubscribe and resubscribe to a topic often without crashing.

#### Actual behavior

When unsubscribing and resubscribing to a topic during a runtime reset, we're seeing crashing happen frequently. See ticket https://github.com/ros-planning/navigation2/issues/1363

#### Additional information

dirk-thomas, 2019-12-10:
Can you please provide reproducible steps (as minimal as possible, so without navigation would be preferred but isn't a requirement).

SteveMacenski, 2019-12-10:
I'm just the messenger. @crdelsey and @mlherd  report it. But I dont think from our discussions its easy to reproduce at the application layer, but it happens over a short time horizon.

crdelsey, 2019-12-10:
I don't have an easy test case implemented at the moment. However, I believe the core of the failure sequence goes something like this:

1. There is must be no publisher for PointCloud2 messages in the system.
2. Create a subscriber to PointCloud2 and let the system settle.
3. Delete the subscriber object and then immediately create a new one on the same node.

In our case, there are a lot of other subscribers and publishers as well as other traffic. I'm not sure at this point how much of that stuff is important to reproduce the problem.

dirk-thomas, 2019-12-10:
> I'm not sure at this point how much of that stuff is important to reproduce the problem.

I would suggest to try coming up with a trivial example (e.g. modifying the demo `listener` to destroy and recreate) to check if that can reproduce the problem.

crdelsey, 2019-12-11:
Well, the simple test didn't do it. It probably won't be till next year when I have time to dig into this in more detail

dirk-thomas, 2020-01-16:
@crdelsey Friendly ping.

crdelsey, 2020-01-16:
I'll try to reproduce this in a few days. I need to get nav2 back to a working state first.

Since this is unlikely to be easy to reproduce, is there any particular information that you'd want to see as a starting point? Wireshark logs? I could set a breakpoint at the line where the error message occurs and provide a core dump.

sainimayankmohan, 2020-08-07:
Hi , I am also getting this error when i am creating subscription to clock message or sensor data image message. It is random and doesn't occur every time. I am resetting a subscription shared pointer to unsubscribe and than subscribe again.

(gdb) bt
#0  0x0000000000000061 in ?? ()
#1  0x00007ffff4d0d65f in rmw_create_subscription () from /home/msaini1/lib/librmw_fastrtps_cpp.so
#2  0x00007ffff739dd72 in rcl_subscription_init () from /home/msaini1/lib/librcl.so
3  0x00007ffff7ad44f9 in rclcpp::SubscriptionBase::SubscriptionBase(std::shared_ptr<rcl_node_t>, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t const&, bool) () from /home/msaini1/lib/librclcpp.so
#4  0x00000000004b2970 in rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >::Subscription(std::shared_ptr<rcl_node_t>, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t const&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >) ()
#5  0x00000000004b232c in void __gnu_cxx::new_allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >::construct<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, std::shared_ptr<rcl_node_t>, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&>(rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >*, std::shared_ptr<rcl_node_t>&&, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&) ()
#6  0x00000000004b16ed in void std::allocator_traits<std::allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > >::construct<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, std::shared_ptr<rcl_node_t>, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&>(std::allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >&, rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >*, std::shared_ptr<rcl_node_t>&&, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&) ()
#7  0x00000000004b0cb7 in std::_Sp_counted_ptr_inplace<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> >, std::allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<std::shared_ptr<rcl_node_t>, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > > const&>(std::allocator<rclcpp::Subscription<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > >, std::shared_ptr<rcl_node_t>&&, rosidl_message_type_support_t const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_subscription_options_t&, rclcpp::AnySubscriptionCallback<rosgraph_msgs::msg::Clock_<std::allocator<void> >, std::allocator<void> > const&, rclcpp::SubscriptionEventCallbacks const&, [... cut ...] d::remove_cv<std::remove_reference<rclcpp::QoS const&>::type>::type>::value, void>::type, rclcpp::subscription_traits::has_message_type<!std::is_base_of<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::remove_cv<std::remove_reference<rclcpp::QoS const&>::type> >::value, void>::type, rclcpp::subscription_traits::has_message_type<!std::is_same<rmw_qos_profile_t, std::remove_cv<std::remove_reference<rclcpp::QoS const&>::type> >::value, void>::type>::type,

clalancette, 2024-01-03:
Given the age of this, and the fact that it couldn't reproduce later on, I'm going to close this out.  If this is still a problem, please feel free to reopen.

closed 2024-01-03 by clalancette