ros2/rmw_fastrtps #352: remote nodes communicate with long time delaying

aitazhixin, 2020-03-09:
When i test remote node communication, two PCs are connected by network cable. PC-A as publisher, and PC-B as subscriber. ROS2 msg type is sensor_msgs/msg/Image, and image size is 720 * 1280. Msg delay reaches from 80ms to 140ms. If publisher and subscriber run on the same PC, msg delay just 1ms or 12ms. Is there any idea to impove transportation of image sensor data in ROS2?

I do not install others rmw, only default fastrtps.
`ii  ros-eloquent-fastrtps                               1.9.3-2bionic.20200218.174738                       amd64        Implementation of RTPS standard.
ii  ros-eloquent-fastrtps-cmake-module                  0.8.0-1bionic.20191213.042825                       amd64        Provide CMake module to find eProsima FastRTPS.
ii  ros-eloquent-rmw-fastrtps-cpp                       0.8.1-1bionic.20200218.175817                       amd64        Implement the ROS middleware interface using eProsima FastRTPS static code generation in C++.
ii  ros-eloquent-rmw-fastrtps-shared-cpp                0.8.1-1bionic.20200218.175556                       amd64        Code shared on static and dynamic type support of rmw_fastrtps`

jacobperron, 2020-03-19:
This sounds like it may be related to https://github.com/eProsima/Fast-RTPS/issues/1062. I'm experiencing delays (and message drops) on the same machine when load is high.

@MiguelCompany FYI. Do you think we should close this as a duplicate or keep it as a separate ticket?

MiguelCompany, 2020-03-20:
As it refers to eloquent, it is more likely to be a duplicate of #338, even though [eProsima/Fast-RTPS#1062](https://github.com/eProsima/Fast-RTPS/issues/1062) may also be related.

Floria-Xu, 2022-05-02:
Hey! I'm also testing the performance of the image message. However, I'm just a beginner of ROS2. May I consult you how you test latency between two nodes? I can only minus the time of two nodes in logging and I wonder if there is a easier and more direct way to get the latency.Thank u in advance!