ros2/rmw_fastrtps #356: Add support for Support for ON_REQUESTED/OFFERED_INCOMPATIBLE_QOS events

mm318, 2020-03-27:
----
## Feature request

#### Feature description

Add support for ON_REQUESTED_INCOMPATIBLE_QOS and ON_OFFERED_INCOMPATIBLE_QOS events. This will enable `rclcpp`/`rclpy` make client callbacks regarding incompatible QoS events, which are when participants on the same topic are discovered but are unable to connect due to QoS policy incompatibilities.

This is required to complete feature request https://github.com/ros2/ros2/issues/822.

#### Implementation considerations

Extend `PubListener` and `SubListener` to respond to `on_requested_incompatible_qos` and `on_offered_incompatible_qos` callbacks respectively. Also, make the relevant changes to guard conditions for events and `takeNextEvent()`.

However, a successful implementation of this feature is blocked by https://github.com/eProsima/Fast-RTPS/issues/852.

ivanpauno, 2020-03-27:
@MiguelCompany FYI, ROS2 added support to these callbacks.
For `fastrtps` based implementations, if an user creates one of these callbacks it gets a `NonImplemented` error.

MiguelCompany, 2022-02-18:
@mm318 @ivanpauno see #583

PR #583 (2022-02-22): Complete events support
This replaces #581 and adds support for the three missing events:

* **RMW_EVENT_OFFERED_QOS_INCOMPATIBLE**: The interfaces are present and working on Fast DDS.
* **RMW_EVENT_REQUESTED_QOS_INCOMPATIBLE**: The interfaces are present and working on Fast DDS.
* **RMW_EVENT_MESSAGE_LOST**: The interfaces are present in Fast DDS, but the listener is currently never called.
  This means the RMW will behave as if messages are never lost, until we implement on_sample_lost on Fast DDS v2.6.0.

Connects to ros2/rcl#960, ros2/rclcpp#1887, and ros2/rclpy#895

Closes #356
closed 2022-02-22 by clalancette